MicrosoftDocs/windows-driver-docs #2890: InfVerif Task information needed

MrDavid949, 2021-09-08:
[Enter feedback here]
Hi, can you please add a section here for the InfVerif task?  Specifically how to invoke infverif.exe with different command line options. Changing settings to the InfVerif task via MsBuild is not documented anywhere. The existing sections for wmimofck, stampinf, and mofcomp tasks are good.

thank
you

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1fd8158a-768a-6edc-2ced-0a9da038f6c9
* Version Independent ID: 44220be0-e804-4a83-93a4-5cb1dd61ecb4
* Content: [Wmimofck task - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/wmimofck-task)
* Content Source: [windows-driver-docs-pr/devtest/wmimofck-task.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/wmimofck-task.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2021-09-15:
Hi @MrDavid949,

MSBuild runs infverif as a task automatically, right after stampinf.  It uses the "Target Platform" in the project properties (**Configuration Properties->Driver Settings**) to auto-configure as follows:

Target Platform = Desktop. InfVerif /k
Target Platform = Universal. InfVerif /u
Target Platform = Windows Driver. InfVerif /w

We don't support changing that automatic invocation.  If you want to run infverif as a standalone, you can configure a bit more.  Those options are described on https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/running-infverif-from-the-command-line.

Does that help?

MrDavid949, 2021-09-15:
Hi, since I'm building a driver I think I need to keep the target platform set to "Windows Driver" which seems to lock me in to the /w option. I also found that page describing the command line.

I have created a custom target in my .vcxproj and called InfVerif.exe with the options I wanted.  It's not the best solution since I'm basically running InfVerif.exe twice during my build, once via the InfVerif task with /w and once in my custom step with the options I want. I guess there is no other way.

tedhudek, 2021-09-17:
That sounds right to me.  I went ahead and added a short section with the info above in case someone else has the same question.  Thanks again for caring about the docs!

https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/running-infverif-from-the-command-line#infverif-and-msbuild-tasks

closed 2021-09-17 by tedhudek