MicrosoftDocs/windows-driver-docs #2898: Outdated tools, instructions lacking & hard to implement.

Tamire, 2021-09-13:
To whom it may concern,
I have been having a difficult time understanding and creating an extension unit DLL.
firstly, the tools are outdated:
You are requesting installation of toolset published in 2003, without any download links.

It would have been nice to have a ready-to-go project which could have been built on VS2019.
How could one build a new UVC Extension unit without this dll?
Even after finding the outdated build tools/environment I am getting linker errors which I cannot find much information about online:
error LNK1181: cannot open input file 'objfre_wxp_x86\i386\uvcxuplgn.obj'

Any help or update on the issue would be highly appreciated.

Best regards,
Tamir.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9d55cadd-9e44-7200-4e65-efd0ca9ead38
* Version Independent ID: 7179ce40-024c-4b8e-3191-4a7ca7798f88
* Content: [Building the Extension Unit Sample Control - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/building-the-extension-unit-sample-control)
* Content Source: [windows-driver-docs-pr/stream/building-the-extension-unit-sample-control.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/stream/building-the-extension-unit-sample-control.md)
* Product: **windows-hardware**
* Technology: **stream**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-09-17:
Thanks for bringing this outdated content to our attention, @Tamire.

The content and sample code is currently unsupported and may not work with the current toolchain.  We are tracking this issue internally but have nothing further to add at this time.

closed 2021-09-17 by barrygolden

nicedayzhu, 2022-03-20:
error LNK1181: cannot open input file 'uvcxuplgn.obj'
The same error happend to me either, still no sollution