MicrosoftDocs/windows-driver-docs #2900: I don't see a clear link to Windows 10 drivers for installation on the driver page - can you direct me specifically to this

got879, 2021-09-14:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a741c2c3-ee16-97ea-94b0-10f0df8377b9
* Version Independent ID: b8acde13-4a58-4c39-7525-e175257a44de
* Content: [Overview of Remote NDIS (RNDIS) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/overview-of-remote-ndis--rndis-)
* Content Source: [windows-driver-docs-pr/network/overview-of-remote-ndis--rndis-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/overview-of-remote-ndis--rndis-.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-09-14:
Hi @got879, have you tried the directions on [Introduction to Remote NDIS (RNDIS)](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/remote-ndis--rndis-2)? Microsoft Windows versions beginning with Windows XP include a Remote NDIS driver for USB devices. This NDIS miniport driver, Rndismp.sys, is implemented and maintained by Microsoft and is distributed as part of all supported Windows versions. You can find it in the %SystemRoot%\System32\drivers directory.

got879, 2021-09-14:
Is it possible to install just this driver. It is not present in the list of devices and does not install when a Remote NDIS device is connected via USB.

aviviano, 2021-09-17:
Hi @got879,
The current documentation on how to access the driver is correct. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation. 
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive. Best of luck with this issue.
* [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

closed 2021-09-17 by aviviano