MicrosoftDocs/windows-driver-docs #2907: Submit new hardware missing

bsbaixo, 2021-09-17:
The very first step on the page is to click on Submit New Hardware on my partner page.
That option isn't present.

Is there an alternative route?

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 16149520-06cf-5257-0e53-0aeda1719b2a
* Version Independent ID: 63b8a418-0c66-c9f9-321f-f954543ec452
* Content: [Create a new hardware submission - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-hardware-submission)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-hardware-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-hardware-submission.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2021-09-17:
Hi @bsbaixo

Is the account you're using to create a new submission registered in the Partner Center Hardware program?

 ~ *Mark*

bsbaixo, 2021-09-18:
Hi,

I don't think so, but we don't develop hardware. It wasn't clear if it was
needed for non hardware drivers.
Is that program the only way to sign a driver, hardware or not?

Thanks,
B

mhopkins-msft, 2021-09-20:
Hi @bsbaixo,

It sounds like you have questions about the driver certification process. This GitHub issues mechanism is for submitting feedback about the documentation. I recommend that you submit a question on the [Partner Center Support](https://partner.microsoft.com/support/) page.

You could also check the [Hardware dashboard FAQ](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq) for more options.

I'm going to close this issue since it doesn't directly pertain to the documentation.

I wish you the best,
 ~ *Mark*

closed 2021-09-20 by mhopkins-msft