MicrosoftDocs/windows-driver-docs #2910: AddSoftware Directive flags interpretation

thxkiwi, 2021-09-21:
I'm trying to debug a possible INF issue and I'm trying to properly understand the flags used for the AddSoftware directive. 

When the flags are 0x00000000 the AddSoftware directive is processed only once. What is the scope of this? 
* Once for life of the Windows install (essentially it will only ever run the first time since the install of Windows)
* Once for scope of this INF?
* Once for the life of this INF as long as the INF is installed?
* Something else?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ffd84f19-a70c-f319-05d7-de54a24c90b7
* Version Independent ID: 08e48321-ba5b-6204-f795-b1e42cd093f6
* Content: [INF AddSoftware Directive - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addsoftware-directive)
* Content Source: [windows-driver-docs-pr/install/inf-addsoftware-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-addsoftware-directive.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-09-22:
Hi @thxkiwi,

Thanks for the question. The flags determine if the software needs to be installed once globally, or it needs to be installed once per device instance for which the INF is applicable.

Say there are three devices are installed with the same INF -- the software installation process runs only once for flags 0x00000000 but three times for flags 0x00000001.

I believe once it's installed, the software is there for the duration of the Windows installation - although it would reset on OS upgrade.

I added a line to the doc page to reflect the above - it should appear on the live site within a few minutes.

Hope this helps, and best of luck!

closed 2021-09-22 by tedhudek

thxkiwi, 2021-09-22:
Thanks @tedhudek.

I see multiple "scopes" here;
1. "Global", regardless of SoftwareName
1. Within the INF, regardless of SoftwareName (but then does that make it simple the instance of AddSoftware on that line number in that INF?)
1. SoftwareName - globally. 
1. SoftwareName - within the INF

My confusion is how SoftwareName plays into this with each of the different values of flags.

Some cases
1. flags = 0; SoftwareName is used in two INFs
2. flags = 0; SoftwareName is used on multiple different lines of the same INF
3. flags = 1; SoftwareName is used in two INFs
4. flags = 1; SoftwareName is used on multiple different lines of the same INF

Some other things I'd like to confirm;
1. The naming of the DDInstall does not factor
2. The naming of the software-install-section does not factor

I've tried testing these scenarios and it's hard to observe without first knowing factors involved and the scope in which they operate.

Thanks for helping to clarify my understanding.

tedhudek, 2021-09-23:
Thanks for the clarification @thxkiwi, I will investigate further and see what I can find for you.

tedhudek, 2021-09-28:
Update - haven't learned anything new on this so far.  Will let you know if I do.

andylsn, 2021-09-29:
@thxkiwi for now, SoftwareName should be system-wide unique regardless of the INF or section name. That's why we suggest using a unique identifier for SoftwareName. Think of it as similar global state as AddService.

> We recommend prefacing the SoftwareName with the vendor name, for example ContosoControlPanel

SoftwareName is mainly for version check so that a lower or same version doesn't get processed. The version info is either stored in system state or device state. Flags=0 means the check is performed on system level while Flags=1 means the check is performed on every device separately. 

>The processing of an AddSoftware directive checks the version against previous software installed with the same name by an AddSoftware directive from any driver package. 

It's worth noting that uninstallation is a totally different process. For now, driver/device uninstallation has nothing to do with software uninstallation.

>Any software installed using AddSoftware will not be uninstalled if the virtual software component device or its parent devices are uninstalled.

Please let me know if you have any questions.