MicrosoftDocs/windows-driver-docs #2914: Missing Windows 11 values

superm1, 2021-09-23:
Are there new strings introduced for Windows 11?  If so, can they please be added to this table.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7eea1fbb-7f7d-d32d-da26-07f6499b3336
* Version Independent ID: 3fed902a-fb51-cd76-29d1-3fdaa97d847f
* Content: [How to Identify the Windows Version in ACPI by Using _OSI - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/acpi/winacpi-osi)
* Content Source: [windows-driver-docs-pr/acpi/winacpi-osi.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/acpi/winacpi-osi.md)
* Product: **windows-hardware**
* Technology: **acpi**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-09-23:
Thanks for submitting this issue, @superm1.

I've checked in with the ACPI engineering team. The _OSI string for the next version of Windows is work in progress and has not yet been released to documentation for publication. I will update this topic as soon as I receive the information from the ACPI team.

barrygolden, 2021-09-23:
Closing this issue as no additional action is possible at this time. Will update the issue after the _OSI string for the next version of Windows has been added to this topic.

closed 2021-09-23 by barrygolden