MicrosoftDocs/windows-driver-docs #2917: Can't find documentation for SRCSRV version 3

mstange, 2021-09-26:
[The page for the SrcSrv language specification version 1](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/language-specification-1) says about the `VERSION` variable:

> Current versions of SrcSrv use a value of 2.

There is also [a short page documenting version 2](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/language-specification-2).

However, it seems that there exists also a version 3. This is evidenced by [this code from PerfView](https://github.com/microsoft/perfview/blob/5c9f6059f54db41b4ac5c4fc8f57261779634489/src/TraceEvent/Symbols/NativeSymbolModule.cs#L778):

```cs
            ///  ********* This is a typical example of what is in a PDB with source server information. 
            ///  SRCSRV: ini ------------------------------------------------
            ///  VERSION=3
            ///  INDEXVERSION=2
[...]
```

But I can't find any docs for version 3.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ae6e0df1-4638-9133-3315-23352641b625
* Version Independent ID: 314c937e-aad2-90c0-4eef-5bb557c7c8e3
* Content: [Language Specification 1 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/language-specification-1)
* Content Source: [windows-driver-docs-pr/debugger/language-specification-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/language-specification-1.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-09-28:
@mstange - Thanks for pointing this out. I checked with dev team, and although there is a version 3, it remains unchanged from V2. I added a small V3 topic to document this. Good luck with your source server development work.

closed 2021-09-28 by DOMARS

mstange, 2021-09-28:
Thank you so much!