MicrosoftDocs/windows-driver-docs #2919: Typo: VERCTL should be VERCTRL

mstange, 2021-09-27:
In the example listing, the page lists:

```
VERCTRL=<source_control_str>
```

but then in the part below, where each variable is explained, it says `VERCTL` without the R.

The R should be present.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ae6e0df1-4638-9133-3315-23352641b625
* Version Independent ID: 314c937e-aad2-90c0-4eef-5bb557c7c8e3
* Content: [Language Specification 1 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/language-specification-1)
* Content Source: [windows-driver-docs-pr/debugger/language-specification-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/language-specification-1.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-09-27:
@mstange - Yes, I checked and it should be VERCTRL with the R. Topic is updated to fix this typo. Thanks for spotting this and creating this issue. You are also welcome to create PRs for fixes like this. Thanks again.

closed 2021-09-27 by DOMARS

mstange, 2021-09-28:
Thanks! In the future I'll submit PRs for similar issues.