MicrosoftDocs/windows-driver-docs #2922: url in Manufacturer info

FrederikZTS, 2021-09-28:
[Enter feedback here]
It is not clear if we are allowed to enter an url besides the company name. Are there any limits (besides length) to what we enter as manufacturer string?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9445c977-9f8c-00c6-68fa-b575ab325fcf
* Version Independent ID: 58963a0b-6f4b-b6ee-f454-7e72e68d7bef
* Content: [INF Manufacturer Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section#examples)
* Content Source: [windows-driver-docs-pr/install/inf-manufacturer-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-manufacturer-section.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-09-28:
Do you mean like "Contoso" below (for which you would also have a section of the same name) or "Contoso, Ltd." below, meaning the string assigned to the %strkey% token?  I'm guessing you mean the former but wanted to check.

```
[Manufacturer]
%Mfg%=Contoso        ; Models section == Contoso

[Contoso]

; ...
[Strings]
Mfg = "Contoso, Ltd."
```

FrederikZTS, 2021-09-29:
I mean the string, like "Contoso, Ltd." 
Would "Contoso, Ltd. (https://www.Contosos.com)" also be an acceptable manufacturer string?
As it will be displayed in the general tab of the driver properties (when selected through device manager)

tedhudek, 2021-09-29:
I'm pretty sure this will work fine, although it wouldn't be clickable anywhere.  Thanks for the question, and good luck with your implementation!

closed 2021-09-29 by tedhudek