MicrosoftDocs/windows-driver-docs #2923: Operating System Codes Missing in Documentation

Ben-Carpenter, 2021-09-28:
In Partner Center UI there is an option for MN when targeting a new shipping label. This option is not listed in the `List of Operating System Family Codes` or `List of Operating System Codes`.  Is there documentation elsewhere for MN ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1c20769b-d969-a51a-4eb3-f9cc27875f28
* Version Independent ID: 853fe2a9-d654-98a3-7635-c30a450cdbdc
* Content: [Get product data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#list-of-operating-system-codes)
* Content Source: [windows-driver-docs-pr/dashboard/get-product-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-product-data.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2021-10-04:
Hi @Ben-Carpenter

I believe "MN" is for Manganese. I am trying to find more information and will update when I have more to share.

Thanks,
 ~ *Mark*

mhopkins-msft, 2021-10-05:
Hi @Ben-Carpenter

I've verified that MN represents the Manganese release, also known as Windows 10X. That release was cancelled. However, the MN option has not yet been removed from the portal because there are higher priority work items that the team is addressing first. Sorry for the confusion.

Thanks,
 ~ *Mark*

closed 2021-10-05 by mhopkins-msft