MicrosoftDocs/windows-driver-docs #2931: For `symstore-command-line-options` Does `-:NOREFS` require `/p` to be used?

seanbudd, 2021-10-03:
The usage guide, specifically these lines, seem to imply that `-:NOREFS` requires `/p`, as the syntax used is `[/p ... [-:NOREFS]]`.

```ps1
symstore add [/r] [/p [/l] [-:MSG Message] [-:REL] [-:NOREFS]] /f File /s Store /t Product [/v Version] [/o] [/c Comment] [/d LogFile] [/compress]
symstore add [/r] [/p [/l] [-:REL] [-:NOREFS]] /g Share /f File /x IndexFile [/a] [/o] [/d LogFile] 
symstore add /y IndexFile /g Share /s Store [/p [-:MSG Message] [-:REL] [-:NOREFS]] /t Product [/v Version] [/o] [/c Comment] [/d LogFile] [/compress]
```

However further in the documentation, the requirement for `/p` is not mentioned under `-:NOREFS`, even though that is mentioned for `/l`, `-:MSG`. and `-:REL`.

Additionally, usage such as `symstore add /s symbols /compress -:NOREFS /t {product} /f {file}` worked up until now, instead throwing the following error:

`SYMSTORE ERROR: Class: Syntax. Desc: Invalid compression type '-:NOREFS'`

PS: The example usage was using [this windows image](https://www.appveyor.com/docs/windows-images-software/#visual-studio-2019)..

#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 561f2c34-2b98-1b84-a94a-8e580133a96e
* Version Independent ID: ae25edff-62f7-8659-40a1-4ff6f3e842f6
* Content: [SymStore Command-Line Options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symstore-command-line-options)
* Content Source: [windows-driver-docs-pr/debugger/symstore-command-line-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symstore-command-line-options.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

seanbudd, 2021-10-04:
As noted in https://github.com/nvaccess/nvda/pull/12903#issuecomment-933225526, the usage guide is inconsistent between these documents

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symstore-command-line-options
As opposed to the docs at:
https://docs.microsoft.com/en-us/windows/win32/debug/symstore-command-line-options

DOMARS, 2021-10-05:
@seanbudd - Hello, I'm looking at this issue and it does look like /p maybe required per the syntax definition brackets. 

On the use of the compress option, it might be more clear to specify the type of compression used for example `/compress CAB`

On a tangential note, for large projects you may also want to be aware of that there is a devops pipeline process that will create symbols. https://docs.microsoft.com/en-us/azure/devops/pipelines/artifacts/symbols?view=azure-devops

DOMARS, 2021-10-06:
@seanbudd - I updated the topic to mention that NOREFS is used with the /p option for pointer file creation. And you may be aware, but compress is not used with pointer files, so /p and /compress would not be used together. I also added the ZIP|CAB option for compress that is shown in the command line help. 

I should mention, that for other information on the use of symbols, there is also a discussion area that driver developers use to figure out challenging issues with building code. [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html). You can post your question there, and the community of developers usually give feedback fairly quickly. Good luck with your use of the symstore command in your build process.

closed 2021-10-06 by DOMARS

seanbudd, 2021-10-06:
Thanks @DOMARS.