MicrosoftDocs/windows-driver-docs #2932: Different "unknown" values for RSCP in serving and neighboring cells

aleksander0m, 2021-10-06:
The RSCP field (int32) in the MBIM_UMTS_SERVING_CELL_INFO type says:

The Received Signal Code Power of the serving cell. The range is -120 to -25, in units of 1dBm. Use 0 when this information is not available.

The RSCP field (int32) in the MBIM_UMTS_MRL_INFO type says:

The Received Signal Code Power of the serving cell. The range is -120 to -25, in units of 1dBm. Use 0xFFFFFFFF when this information is not available.

Shouldn't the 2nd one say "0 when this information is not available"?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 05ac7540-198a-9a9a-c8cd-1bb4eac81242
* Version Independent ID: 5223dc75-17e7-d9cd-94dd-d21704fa331c
* Content: [MB base stations information query support - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-base-stations-information-query-support)
* Content Source: [windows-driver-docs-pr/network/mb-base-stations-information-query-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-base-stations-information-query-support.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-12-02:
Thanks for opening this issue @aleksander0m. We've looked into these values, and updated MBIM_UMTS_MRL_INFO to:
"Use **0** when this information is not available."
Please open a new issue if you have further questions. 
Thanks again.

closed 2021-12-02 by aviviano

aviviano, 2022-01-07:
Hi @aleksander0m, I understand there is inconsistency. However, we reviewed all of the RSCP values and current documentation is correct for TDSCDMA .