MicrosoftDocs/windows-driver-docs #2956: ExAllocatePoolZero

lousybrit, 2021-11-02:
[Enter feedback here]

It does not cover any or all the NDIS or StorPort memory allocation functions and if they are going to zero buffers or not.
Should these calls be combined with RtlZeroMemory calls? Are they going to zero memory automatically from Windows 10 2004 releases onwards?
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ba2a6e9f-8e86-c441-c9cc-bf9c3718d175
* Version Independent ID: 607d12cd-80a2-ff35-d538-330f785e2617
* Content: [WdkDeprecatedApi (Windows Driver CodeQL Query) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/codeql-windows-driver-wdkdeprecatedapi)
* Content Source: [windows-driver-docs-pr/devtest/codeql-windows-driver-wdkdeprecatedapi.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/codeql-windows-driver-wdkdeprecatedapi.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-11-08:
@lousybrit - I have reached out to the dev team for additional information on updated memory allocation for NDIS and Storport and the idea of expanded CodeQL tests.  For an example DDI for NDIS, I see [NdisAllocateSharedMemory function (ndis.h)](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ndis/nf-ndis-ndisallocatesharedmemory). To make sure I'm understanding your input, are there other NDIS Memory allocation DDIs, that you are considering? And do you have an example  memory allocation call for Storport?

lousybrit, 2021-11-09:
We have an NDIS Miniport driver, and it uses the following:
NdisAllocateMemoryWithTagPriority
NdisMAllocateSharedMemoryAsync
NdisMAllocateSharedMemoryAsyncEx

Are these routines being deprecated or mapping to functions like ExAllocatePool2 or ExAllocatePoolZero?
So will the Ndis memory allocations return zeroed memory or not?
Should the driver owners be explicitly calling RltZeroMemory or not? In cases where they do such a thing, is that leading to buffers being zeroed twice on newer Windows OS versions?

It would be good to know what is the status of these routines.
Also, does the zeroing of allocated memory apply to the StorPort allocation routines?

Thanks
Steve

DOMARS, 2021-12-15:
@lousybrit - Thanks for this PR. The dev team double checked and the popular current day APIs for NDIS and Storport, such as NdisAllocateMemoryWithTagPriority, NdisMAllocateSharedMemoryAsyncEx and StorPortAllocatePool function do zero out memory. So at this point, they will not be expanding the CodeQL query. But is possible that there may be more APIs that will be added, if any that are in active use, are later surfaced. I think you may have mentioned a deprecated version of an API, those are not recommended for use for a number of reasons, such as test coverage and security updates. The recommendation would be to move as quickly as possible to the currently supported memory allocate DDIs. Thanks again for this PR, and your interest in helping everyone to continue to increase the security of their memory allocation code.

closed 2021-12-15 by DOMARS