MicrosoftDocs/windows-driver-docs #2958: WDTF simple I/O action plugin project ... where is it?

IC-SteveP, 2021-11-04:
Hi.  Please tell me where I can fine this.  I have installed VS2015, VS2017 and VS2019 and many WDKs and haven't seen what is described on this page...

> From the File menu, click New > Project.
> From the list of installed templates in the New Project dialog box, select Visual C++ > Windows Driver > Tests > WDTF Simple I/O Action Plug-in.
> 

The devices I'm working with require a custom WDTF_SIMPLEIO plug-in to run HLK tests for WHQL so our customers can run our drivers now that MS has changed the rules for Server 2016 and up.  So, like, kinda freakin' important, eh.

Regards,
    Steve
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: efa2458c-9f17-37e0-9ba7-21e3df5ba820
* Version Independent ID: 912f570e-b93f-f2ef-1436-f6ebb5e8e65a
* Content: [Customize device I/O using a WDTF simple I/O action plug-in - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdtf/to-customize-i-o-for-your-device-using-the-wdtf-simple-i-o-action-plug-in)
* Content Source: [windows-driver-docs-pr/wdtf/to-customize-i-o-for-your-device-using-the-wdtf-simple-i-o-action-plug-in.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdtf/to-customize-i-o-for-your-device-using-the-wdtf-simple-i-o-action-plug-in.md)
* Product: **windows-hardware**
* Technology: **wdtf**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

IC-SteveP, 2021-11-05:
Apologies for my snarkyness at the end there ...

DOMARS, 2021-11-09:
IC-SteveP – Thanks for pointing out that the I/O plugin option is no longer available in the latest version of the kits. I’m working with dev team to create updates for these topics. As that work starts, here a few ideas that may help with your current work.

In the current kit, there is some older example code, you could look at – ‘C:\Program Files (x86)\Windows Kits\10\Testing\Samples\WDTF`. As these are older samples, you may need to update the include file references to get them to work with the latest versions of Windows, if that is possible.

An older version of the WDK such as 8.1 when matched with the Visual Studio  may contain the plugin. For information on downloading older versions of the Visual Studio and the matching WDK is available here : https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads.  I don't have information on which version of the WDK has the VS install piece that  creates this plug-in. Also some of the older versions of the WDK will overwrite and don't support side by side installation.

I’m not sure if either of these options will be helpful, but wanted to pass on what I have right now, as the doc updates may take some time to complete.

DOMARS, 2021-11-12:
@IC-SteveP - I have contacted the dev team and they confirmed that the Simple IO plugin is no longer available and other updates need to be made. This will take some period of time, likely n number of months. I will track this work, and update the docs when those WDK code updates are available. 

The other areas you could check for possible Simple IO samples would be the WDK samples:

https://docs.microsoft.com/en-us/samples/browse/?expanded=windows&products=windows-wdk

As well as possibly in the DMF - Driver Module Framework code set:

https://github.com/Microsoft/DMF/blob/master/README.md

Thanks again for bringing this issue to our attention, and could luck with your Windows Server driver development work.

closed 2021-11-12 by DOMARS

IC-SteveP, 2021-11-22:
@DOMARS  - I look forward to up-to-date and correct docs.

Regards,
    Steve

FYI - Neither of those links point at relevant info.