MicrosoftDocs/windows-driver-docs #2960: "The latter page contains an example"

thxkiwi, 2021-11-08:
The section "Steps in the app" ends with "The latter page contains an example" but I cannot find any latter page with any such example. Please provide a link to the example.

Specifically, I want to add the `DriverDependency` but doing so by directly adding it to the `AppManifest.xml` causes build errors in debug when Visual Studio automatically adds `PackageDependency` entries *after* the manually added `DriverDependency`.

A reference example would be a great help.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d63fca17-2476-874f-5329-5f065552ddad
* Version Independent ID: c5470d95-0ead-9d43-1e6c-04af1acd79bc
* Content: [Pairing a driver with a Universal Windows Platform (UWP) app - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/pairing-app-and-driver-versions)
* Content Source: [windows-driver-docs-pr/install/pairing-app-and-driver-versions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/pairing-app-and-driver-versions.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-11-18:
Hi @thxkiwi, sorry about the delay. Does the example at https://docs.microsoft.com/en-us/uwp/schemas/appxpackage/uapmanifestschema/element-uap5-driverconstraint#examples help you at all?

thxkiwi, 2021-11-18:
@tedhudek it does. I found the issues I was experiencing was an earlier version of Visual Studio (16.6.2), when building a debug build, was placing the `<PackageDependency>` elements for debug dependencies after the `<uap5:DriverDependency>`. This caused the manifest to not validate against the schema. Seems to have been remedied in later versions of Visual Studio, at least it was fixed by 16.11.5. 

I now understand how "the latter page" is used on the page. Closing this ticket with gratitude.

closed 2021-11-18 by thxkiwi

tedhudek, 2021-11-18:
Terrific @thxkiwi, glad to hear you are unblocked.  I will update the verbiage on the page to reflect the VS versioning and to clarify the "latter page" reference, which I agree isn't the clearest :)

tedhudek, 2021-11-18:
Ok, https://docs.microsoft.com/en-us/windows-hardware/drivers/install/pairing-app-and-driver-versions#steps-in-the-app now has these updates.  Thanks again for the feedback @thxkiwi!