MicrosoftDocs/windows-driver-docs #2961: ProcDump Bit Architecture

JonathonAnderson, 2021-11-09:
According to this page, procdump contains both 63 and 32 bit capabilities in a single executable, procdump.exe. If that's the case, why does the download include procdump.exe and procdump64.exe? What's the appropriate use case for each executable? Do I need both executables side by side for this automatic switching to work correctly?

`ProcDump is a "packed" executable containing both the 32-bit and 64-bit version of application - as such, the same executable is used for both 32-bit and 64-bit. When ProcDump runs, it automatically switches the version, if the version running doesn't match the target process.`

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2892006d-13af-24e6-699f-ad5c2e7dc7a5
* Version Independent ID: 8fa52962-0050-1f7e-9664-d8a48c46d429
* Content: [Enabling Postmortem Debugging - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/enabling-postmortem-debugging)
* Content Source: [windows-driver-docs-pr/debugger/enabling-postmortem-debugging.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/enabling-postmortem-debugging.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

chentiangemalc, 2021-11-18:
In my experience most of the time you want procdump.exe which will then auto extract 64-bit version when attaching to 64-bit process. However some scenarios on 64-bit OS require the 64-bit version procdump64.exe such as 64-bit Windows PE, 64-bit Windows without WoW feature installed, or if security software prevents the unpacking of the 64-bit version

DOMARS, 2021-12-16:
@JonathonAnderson - Thanks for filing this issue. I think things have changed, since this section was created. The best approach would be to check the current procdump tool docs, for the latest guidance. I updated the topic to say that.  

In general, you would run the 32 bit version of the utility on the 32 bit version of Windows and the 64 bit version of the utility on 64 bit version of Windows.

There is a procdump -64 option that is described as follows: *By default ProcDump will capture a 32-bit dump of a 32-bit process when running on 64-bit Windows. This option overrides to create a 64-bit dump. Only use for WOW64 subsystem debugging.*

@chentiangemalc - Thanks for passing on your experience here as well.

closed 2021-12-16 by DOMARS