MicrosoftDocs/windows-driver-docs #2973: PCI Multi-Segment support requires a non-zero Segment value

markyoung2u, 2021-11-16:
PCI Multi-Segment has a Segment value besides Bus,Device,Function - where is this to be found, DevicePropertyBusNumber???

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93712a25-6177-1a6f-7e7e-cfa6100c663f
* Version Independent ID: 2fa7063a-a379-ac98-6209-5918bc257af2
* Content: [Hardware ID - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hardware-ids)
* Content Source: [windows-driver-docs-pr/install/hardware-ids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/hardware-ids.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-11-18:
They can be obtained using device property APIs, using this key:

[DEVPKEY_Device_LocationInfo - Windows drivers | Microsoft Docs
](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-locationinfo)

On single-segment system the string will be in the format of: “PCI bus 1, device 2, function 3”

On multi-segment system will read: “PCI segment 1 bus 2, device 3, function 4”

markyoung2u, 2021-11-18:
Ted, I was also concerned about parsing LocationInfo, so this is good to know and I will use this info to change my code today. I hope this will show up on the web docs eventually...

tedhudek, 2021-11-18:
Glad to help, Mark.  I was actually thinking of adding it to the docs.  Do you have a specific page where you think it should go?

markyoung2u, 2021-11-19:
Hmmn. I have 65 tabs in chrome right now, most searching the docs for Segment support. Since it mostly isn't really there, I'm not sure.

  1.  SPDRP_LOCATION_INFORMATION is defined on https://docs.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdigetdeviceregistrypropertya
That says "The function retrieves a REG_SZ string that contains the hardware location of a device." Which is odd to me as that is:
#define REG_SZ                      ( 1 )   // Unicode nul terminated string
But when I call SetupDiGetDeviceRegistryProperty to parse it, the 4th arg "DWORD Property" is really a DEVPROPTYPE arg and it has different sets if defines, and matches:
#define DEVPROP_TYPE_NULL                       0x00000001  // null property data
So this is how things are going for me, for everything you learn, you find out 5 things you did know and 10 things that you can't figure out.

  1.  https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_pci_slot_number defines PCI_SLOT_NUMBER that has no Segment number and

https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/obtaining-device-configuration-information-at-irql---dispatch-level talks about the B/D/F stuff in the last "Copy" section but there is no struct there for the old B/D/F without the Segment, and if you google PCI_SEGMENT_BUS_NUMBER you get 2 links:

  1.  https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-pci_segment_bus_number and
  2.  https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/miniport/ns-miniport-pci_segment_bus_number but those structs are not in my versions of those files, yet

I'd like to see a page for PCI Multi-Segment support as a topic because while UEFI and PCI specs describe it pretty well, amd I think Linux, windows is behind. The place to start all of this and is in need of more clarity is https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/accessing-pci-device-configuration-space
Mark

[EXTERNAL EMAIL]

Internal Use - Confidential

tedhudek, 2021-11-19:
Thanks Mark.  I totally empathize - definitely know that dynamic.

Would it be helpful if I added just the basic workflow above, maybe in a new section on https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/accessing-pci-device-configuration-space?  I'd like to be sure we preserve the info, and get it search indexed, but I don't have a lot of time to spend on it - nor do I have much PCI knowledge.

I'm happy to do that - or you can drop us a PR as well if you like. (**Edit** button on most Docs pages)

I'll reopen this issue while the work item is pending since I am on vacation tomorrow!

tedhudek, 2021-11-19:
Here's a bit more info Mark -- to access this property we recommend:

From kernel mode:
* IoGetDeviceProperty
* IoGetDevicePropertyData 

From user mode:
* SetupDiGetDeviceProperty
* CM_Get_DevNode_Property

If successful, these return a string that needs to be parsed. If you get `DEVPROP_TYPE_NULL`, the property likely does not exist for the device in question.

markyoung2u, 2021-11-19:
Here's another place that will need to be parsed: "@System32\drivers\pci.sys,#65536;PCI bus %1, device %2, function %3;(131,0,0)" so do you know how that will change for a non-Zero segment? This is for RegOpenKeyEx(HKEY_LOCAL_MACHINE, ... with key SYSTEM\\CurrentControlSet\\Enum\\{deviceId}.

tedhudek, 2021-12-02:
Hi @markyoung2u,

I'd recommend using the supported methods previously outlined.  While these device properties are stored (and accessible) in the registry, that should not be relied on, as mentioned here --

https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hklm-system-currentcontrolset-enum-registry-tree

"The Enum tree is reserved for use by operating system components, and its layout is subject to change. Drivers and user-mode device installation components must use system-supplied functions, such as IoGetDeviceProperty and SetupDiGetDeviceRegistryProperty, to extract information from this tree. Drivers and Windows applications must not access the Enum tree directly."

I'm working on adding the issue from this issue to the docs.

markyoung2u, 2021-12-03:
Thanks. Please let me know when there is updated docs.

closed 2021-12-03 by markyoung2u

tedhudek, 2021-12-06:
Glad to help!  Here's the new page -- https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/obtaining-pci-segment-values

Thanks again for reaching out.

markyoung2u, 2021-12-06:
Ted. I see the new page, thank you. I also see on the left pane several topics list including the new one "Obtaining PCI Segment Values" but in the parent link https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/ the page itself does not have a link to the new info. Is this normal? I don't often read the left side, I kind of expect the page I'm reading to have links that are related

tedhudek, 2021-12-06:
Thanks Mark -- are you suggesting that the "In this section" section of https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/ should have a link to https://docs.microsoft.com/en-us/windows-hardware/drivers/pci/obtaining-pci-segment-values?

markyoung2u, 2021-12-06:
Yes. That would be good

tedhudek, 2021-12-06:
Done!