MicrosoftDocs/windows-driver-docs #2979: WHEAErrorSourceMethods::DisableErrorSourceRtn does not work on Windows 11

hahagu, 2021-11-21:
As said on title, the driver method for WHEA,
`WHEAErrorSourceMethods::DisableErrorSourceRtn`
does not function under Windows 11.

This may be a bug of Windows 11,
or/and this page should reflect that information,
or Windows 11 should be fixed.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 13de6bc8-aa78-9e12-6e62-c243f81401ee
* Version Independent ID: 8c3c50a6-8026-6d6d-c72b-f5892702e6f9
* Content: [Disabling an Error Source - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/disabling-an-error-source)
* Content Source: [windows-driver-docs-pr/whea/disabling-an-error-source.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/disabling-an-error-source.md)
* Product: **windows-hardware**
* Technology: **whea**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

hahagu, 2021-11-21:
An example of a service using this method would be:
https://github.com/mann1x/WHEAService

DOMARS, 2022-02-11:
@hahagu  - Thank you for passing on what you are seeing with DisableErrorSourceRtn not working. As you work on this, a few ideas.

- If the code is returning a "STATUS_NOT_FOUND" it is possible that the error source has been renamed, so you may want to confirm that the error source is still available to disable.

- The Get-WmiObject powershell command can be useful to gather information about any specific PCs error sources. For example:

```ps
Get-WmiObject -list * -Namespace root\default
```

You can also use the Set-WmiInstance to modify settings. This command sets the logging level = 2.

```ps
Set-WmiInstance -Class Win32_WMISetting -Argument @{LoggingLevel=2}
```

https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.management/set-wmiinstance?view=powershell-5.1&preserve-view=true

There is additional information on working with WMI scripts in the Win32 documentation.

https://docs.microsoft.com/en-us/windows/win32/wmisdk/creating-a-wmi-script

- Confirm that the app has appropriate rights and permissions to execute on the target system.

- The sample WMI provider driver code, may provide additional information on accessing wmi data from a kernel driver, if you are familiar with Windows driver development.

https://docs.microsoft.com/en-us/samples/microsoft/windows-driver-samples/sample-kmdf-driver-implementing-a-wmi-data-provider/

- There may be a registry key or other approach to disabling the error source.

- If you believe this is an issue with Windows 11, you can use the "Feedback Hub" in Windows to report that to the Windows team.

This particular channel is monitored by the men and women who write the driver documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to provide support for specific code issues.  Your best bet is to use one of the forums where other driver developers and Microsoft engineers can be found. For example, the [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html).

closed 2022-02-11 by DOMARS

mann1x, 2022-02-13:
I have already reported it on the Hub and the forum, no feedback.
The error sources have not been renamed, there are multiple ways to cross check; in primis the WHEA event log entry which is reporting the ID.

tedhudek, 2022-02-14:
FYI -- I updated https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/disabling-an-error-source, adding a Note at the top that lists the error sources that cannot be disabled.

mann1x, 2022-02-14:
@aadriance 

The error source that is meant to be disabled, and should not be critical, is this one:

ID:  0 Type: 16 State: Started       Description: WheaErrSrcTypeDeviceDriver = 0x10, Device Driver Error Source

The others are being attempted to stop because there's a bug in Windows 10.
Unless the others as well are attempted, but fail, to stop, the 0x10 will keep sending WHEA Events even if it shouldn't.
Not a single stability issue in over a year with this method.

aadriance, 2022-02-14:
You're saying that the WMI interface is reporting the error source type 16 as stopped but you are still getting errors from it? Do you have the WHEA-Logger entries in ETW for the error? And you've checked the time the errors are reported to confirm they aren't arriving before the utility gets to disable it?

Error source type 16 is relatively new, but I'm not aware of any issues with its ability to stop currently.

mann1x, 2022-02-14:
I will try to take some ETW logs.
Yes the Device Drive Error Source keeps sending WHEA events after being stopped.
I'm sure of it cause with most AMD Ryzen processors running at high FCLK, including mine, there's a continuous stream of WHEA 19; more under load but also on idle.
For me it only stops when I also stop another source, I don't remember which one.
But when I released the first version of WHEAService some people complained they would still get errors.
Turns out that for others the source to stop was another one.
That's why by default I try to stop them all, this works for everyone.