MicrosoftDocs/windows-driver-docs #2980: Is windows 10x still a thing?

Gunock, 2021-11-23:
There is a Windows 10x row in the `Windows Drivers vs. Windows Desktop Drivers` table.
As far as I know, Windows 10x is not a thing anymore and some of its parts have been integrated into Windows 11.
Correct me if I'm wrong, but this row is no longer valid.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e7dd9475-4e84-c0b0-a4b0-4fd3379e13ba
* Version Independent ID: f554aafc-0f0f-f2f3-4b1d-9f4cf05b2db8
* Content: [Getting Started with Windows Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-windows-drivers)
* Content Source: [windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md)
* Product: **windows-hardware**
* Technology: **develop**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

DOMARS, 2021-11-23:
Assigning to Ted for further re-direction/triage.

tedhudek, 2021-12-01:
Hi @Gunock, appreciate the note. You are correct!  We've updated the doc page with the latest info --

https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-windows-drivers

closed 2021-12-01 by tedhudek