MicrosoftDocs/windows-driver-docs #2986: DOES NOT WORK

valinet, 2021-12-03:
Why bother writing this if it does not work????? Instructions, besides not being even remotely clear, do not work at all.

How's one supposed to do this in Windows 11...? I am tired of Googling and finding ZERO answers. ZERO. `ntsd --` or `ntsd -d -g -G -o -pd -p PID_OF_CSRSS` spit out a `Cannot debug pid ..., Win32 error 0n5 "Access is denied."` error in kernel-attached WinDbg and that's it. Why the fuck is everything "Access is denied". Everything is denied in this OS, always in the name of security. What's there to disable anymore, disabled everything in Windows Defender, Firewall etc, set up "Win32 subsystem debugging" with gflags...

How does one debug `csrss.exe` effectively in 2021...?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a7372174-488b-bda0-6ac5-b0ad2da8fbcc
* Version Independent ID: 52ff3ce8-6cdf-1500-03eb-ba9e4eb0612c
* Content: [Debugging CSRSS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-csrss)
* Content Source: [windows-driver-docs-pr/debugger/debugging-csrss.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-csrss.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

valinet, 2021-12-03:
Maybe it's the same as #667 but I for the life in me cannot understand why has this been ignored for 4 years almost? How can we connect to either of the 2 `csrss` processes that run in modern Windows 11 builds and debug them...?

Why is a bit of transparency so hard to offer... until regulators and governments jump off like hyenas and threatens Microsoft with all kinds of fines and legal actions, doing the right thing is always a long and slow battle, for some reason hard to understand.

DOMARS, 2021-12-23:
@valinet - I updated this topic to reflect the changes that have occurred in debugging CSRSS. This is described on page 112 of the current version of Windows Internals, where they indicate that because CSRSS is a protected process, user mode debugging is not possible. That text provides more information on protected process and the thinking behind this approach.  It is an increasing challenge to balance the needs of diagnostics and debugging, while providing ever increasing safeguards to the operating system. You may be already aware, but it is always a good idea to check the event logs, to see if the OS or an app, realized it hit an issue and logged information with a time stamp near the time you suspect the issue you are encountering is occurring.  Also, seeing what related processes were running at the time of failure, using tools such as process explorer, may provide some additional information to try and determine the root cause. Thanks for filing this issue to highlight what you are running into with debugging CSRSS.

closed 2021-12-23 by DOMARS