MicrosoftDocs/windows-driver-docs #2988: WDK for Visual Studio 2022 when?

jessy-lua, 2021-12-06:
I can't compile any drivers in VS2022 with current WDK what's doesn't integrate in latest VS. 👎 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-drivers-conceptual-docs**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

narayanpvs, 2021-12-20:
Yes, we're moving to Visual Studio 2022. Unavailability of WDK integration with VS2022 forces us to have one leg in VS2019. We'd really appreciate if this could be done really quickly or at least if a timeline could be provided!!

JinZhuXing, 2021-12-24:
I need WDK for VS2022. When can we use it?
Please tell us a timeline.
We have already moved to VS2022.
Thank you.

TheLifeWorm, 2022-01-04:
I need WDK for VS2022. When can we use it?

tedhudek, 2022-01-07:
Sorry for the delay on this one, folks.  It's known issue internally and we're working on it.  I don't have more info to share at this time, but will update here in this issue when I do.

Thanks for your patience!

JinZhuXing, 2022-01-08:
Thank you.
We are waiting.

KomiMoe, 2022-01-11:
I'm waiting too.

tedhudek, 2022-01-25:
Update -- we're currently targeting early Feb to release a WDK/EWDK that works with VS 2022.  The new kit will be available through the Windows Insider program (WIP).

I'll let you know here when that happens.

narayanpvs, 2022-01-26:
Do you mean this will be a pre-release version and unavailable to general
users of the DDK?

Regards,
Narayan

tedhudek, 2022-01-26:
Hi @narayanpvs, this will be a pre-release version, and WIP is available to the general public.

AzeDonLees, 2022-02-01:
Where we can download it?

KomiMoe, 2022-02-02:
> Where we can download it?

did u find it? i wanna download it too. i want to use vs2022 to coding

tedhudek, 2022-02-16:
This kit is now live.  You can find details here -- https://docs.microsoft.com/en-us/windows-hardware/drivers/installing-preview-versions-wdk

Thank you for your patience!

closed 2022-02-16 by tedhudek

narayanpvs, 2022-02-17:
Thank you for the update!
But you're offering a really tough choice:  Either migrate to VS 2022 or
continue supporting drivers for windows 8.1 and below. I wish we didn't
have to choose between the two.
I really hope you support driver development for windows 8 and above using
the new Kit.

Regards

JinZhuXing, 2022-02-17:
Thank you.
When we get stable version of the WDK?

iujakchu, 2022-02-20:
> Thank you. When we get stable version of the WDK?

等鸡啄完了米,you will get the stable version

IULOVE, 2022-03-10:
> Thank you. When we get stable version of the WDK?

Only support win10 or more to give up on it  只支持win10或以上的系统了

wstaelens, 2022-04-04:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB8020	The build tools for WindowsUserModeDriver10.0 (Platform Toolset = 'WindowsUserModeDriver10.0') cannot be found. To build using the WindowsUserModeDriver10.0 build tools, please install WindowsUserModeDriver10.0 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution".	PrintAndShareV4	C:\Program Files\Microsoft Visual Studio\2022\Preview\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets	440	
```

👀

EatonZ, 2022-06-03:
Looks like a stable version is finally available: https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#download-icon-step-3-install-windows-11-version-22h2-wdk

terraslate, 2022-10-19:
Sadly when i try to debug my minifilter i get

---------------------------
Microsoft Visual Studio
---------------------------
Driver Deployment

An error occurred deploying the driver(s) in the solution.

QueryBuildManagerBusyEx must be called on the UI thread.
---------------------------
OK   
---------------------------