MicrosoftDocs/windows-driver-docs #2990: Add IdenTrust EV Code Signing CA

neymars, 2021-12-06:
[Can you please add "Buy an IdenTrust EV Code Signing Certificate poinring to this URL: https://www.identrust.com/certificates/trustid-ev-code-signing]
This is the IdenTrust Commercial Root CA 1 details also available within CCADB:
https://crt.sh/?id=8955726
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93d3f097-d81b-66c2-4835-d63758687d34
* Version Independent ID: fe9d1cdd-966e-89b5-c66b-60531b0ec655
* Content: [Get a code signing certificate - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate#extended-validation-code-signing-certificates)
* Content Source: [windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

neymars, 2021-12-10:
> Hi @neymars <https://github.com/neymars> ,
>
> I am currently researching updating our docs with this additional certificate provider. I will let you know when I have more information.
>
> Thanks,
> ~ Mark

Hi Mark,

You may want to take the same approach taken on this webpage:  <https://docs.microsoft.com/en-us/microsoftteams/direct-routing-plan#public-trusted-certificate-for-the-sbc> https://docs.microsoft.com/en-us/microsoftteams/direct-routing-plan#public-trusted-certificate-for-the-sbc `Direct Routing SIP interface will only trust certificates that are signed by Certificate Authorities (CAs) that are part of the Microsoft Trusted Root Certificate Program. Make sure that your SBC certificate is signed by a CA that is part of the program and that the Extended Key Usage (EKU) extension of your certificate includes Server Authentication. Learn more: Program Requirements - Microsoft Trusted Root Program Included CA Certificate List < <https://docs.microsoft.com/en-us/microsoftteams/direct-routing-plan#public-trusted-certificate-for-the-sbc> https://docs.microsoft.com/en-us/microsoftteams/direct-routing-plan#public-trusted-certificate-for-the-sbc> `

 

Where instead of listing explicit CA's, it shows the full list of CA's in the CCADB. (Common CA Database) - This way you no longer need to maintain this page for approved CA's as the ones in CCADB have the full details.

Thanks

 

Marco Schambach
Program Manager

 

IdenTrust
5225 Wiley Post Way, 

Salt Lake City, UT 84116-4915

+1-801-384-3553 (Office)

+1-954-608-1344 (Mobile)
***@***.*** ***@***.***> 

 <http://www.identrust.com/> www.IdenTrust.com

 <http://www.hidglobal.com/> www.hidglobal.com

mhopkins-msft, 2021-12-17:
Hi @neymars,

Thank you for these suggestions. 

I'm still working on getting info from the code signing team. I've created a work item to get this page updated.

I will proceed to close this GitHub issue. Please let me know if you have any further info to contribute.

Thanks,
 ~ *Mark*

closed 2021-12-17 by mhopkins-msft

mhopkins-msft, 2022-02-11:
Hi @neymars,

Just wanted to let you know that I've added IdenTrust to the code signing page.

Thanks,
 ~ *Mark*

neymars, 2022-02-11:
Thank you Mark!