MicrosoftDocs/windows-driver-docs #2991: What drivers survive a Windows reset?

treestryder, 2021-12-06:
Could you please clarify if drivers in the Driver Store survive a Windows Reset / Wipe. If only some survive, what is the criteria?

We are fully onboard with the modern mobile device management dream. Managing PCs with Intune and onboard them using Autopilot. These PCs are purchased with a clean install of Windows. For those we are migrating, we install a clean copy. When a PC moves between users or roles, we Wipe / reset it. We also rely on Windows Update to maintain its drivers. Normally, this works great. However, we have recently found a couple new models who's NIC and hard drive drivers are missing after a Windows reset. I am assuming the difference must either be:

- Those that fail had a class of driver pre-installed that does not survive the wipe.
- Or, for those that do work, Windows must have a default general driver.

I have also [asked the same at ServerFault](https://serverfault.com/questions/1085249/what-windows-drivers-survive-a-wipe-reset).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b6563342-7aa5-3a02-b758-5a53dd7eca39
* Version Independent ID: a54a4ad3-5eaa-c387-cd91-32e76c6af277
* Content: [Driver Store - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/driver-store)
* Content Source: [windows-driver-docs-pr/install/driver-store.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/driver-store.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-12-07:
Thanks for reaching out, @treestryder.

Can you be more specific about how you are triggering the Windows Reset / Wipe?  Are you referring to one path, or two?  

Are you referring to the reset path that you initiate here?

[image]

treestryder, 2021-12-07:
We are manage PCs using the "modern" Mobile Device Method method, where a device starts with a clean copy of the OS and (when needed) is returned to a clean copy by performing a factory reset. To reset a PC we use the [Intune Wipe action](https://docs.microsoft.com/en-us/mem/intune/remote-actions/devices-wipe#wipe), without checking the "Retain enrollment state and user account" box.

treestryder, 2021-12-07:
I just found a page with Windows S Mode driver requirements. I wonder if these are what I am looking for, or at least related, just that the name is limiting.
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements

tedhudek, 2021-12-13:
Here's a link that has some info on PBR and restoring drivers -- https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/how-push-button-reset-features-work?view=windows-11#restoring-drivers

tedhudek, 2021-12-13:
That's all I've been able to find for you, @treestryder.  If you're still experiencing issues, you could try one of the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive. Best of luck!

* [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

closed 2021-12-13 by tedhudek