MicrosoftDocs/windows-driver-docs #2993: The build tools for WindowsApplicationForDrivers10.0 cannot be found

swingbozo, 2021-12-08:
[Enter feedback here]

This has to be the single most stupid thing Microsoft has ever done in this history of stupid things. I wanted to compile usbview simply to look at some usb stuff. Well, I can't build it because the three versions of the stupid drivers kit don't seem to flag visual studio 2022's "you have have the WDK installed. So I spend the next hour trying to figure out what I needed to download to get the stupid "you don't have the driver kit 10.0 installed on my system" error to go away. I still don't know how to do this. Here's how to reproduce this problem:

Install visual studio 2022 community. Download the usbview program from github. Try to compile it then try to figure out what stupid versions of what stupid thing you need to download next to get it to to build. This is STUPID. Why this has taken me THREE HOURS AND I STILL DO NOT KNOW WHAT IS WRONG WITH THIS STUPID SETUP. I know I did this with VS2019 and it was a snap because I did this exact same thing barely three months ago.

THIS IS STUPID. I CANNOT COMPILE USBVIEW BECAUSE SOMETHING I HAVE INSTALLED IS APPARENTLY NOT INSTALLED. I can't believe how mind bogglingly incompetent this process is.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-drivers-conceptual-docs**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

swingbozo, 2021-12-08:
I finally found this stupid option.

"Platform Toolset" which seems to default to something that doesn't exist for some unknown reason. I can't believe it took me this long to find. I changed this to "Visual Studio 2022(v143)" and it magically fixed it.

So here's a fix for you. When loading a project DO NOT SELECTED PLATFORM TOOLKIT THAT STATES "WindowsApplicationForDrivers10.0 cannot be found." How about if you pick something stupid like, OH, A PLATFORM TOOLKIT THAT IS ACTUALLY INSTALLED INSTEAD OF ONE THAT ISN'T.

tedhudek, 2021-12-09:
Hi @swingbozo - really sorry to hear about the trouble you had.  We're tracking this internally now and hope to have a fix soon.

In the meantime, we could try updating the docs to spare others what you went through.  I can do that if you can confirm the setup that worked - you have VS 2022 and the Win11 WDK, and you changed the Platform Toolset from the default to Visual Studio 2022(v143)?

Thanks for letting us know about the problem.

swingbozo, 2021-12-09:
Yes, this is correct. It was picking "the latest installed" instead of just
picking the 143 (2022) version. I figured "latest installed" would be good
but it just ended up getting it confused. Thank you for getting back to me.

Tim

tedhudek, 2022-01-06:
Hi Tim, thanks again for sharing your workaround.  I asked around and couldn't get a guarantee that this workaround will work in all cases to enable VS2022 to work with the Win11 WDK, so I'm going to close this issue out without updating the docs.

Happy New Year!

Ted

closed 2022-01-06 by tedhudek

zyzhucreol, 2022-08-19:
Thanks for sharing this walkaround! I encountered the exactly same issue when building a project targeting pcie devices from Xilinx, despite Windows Driver Kit 10.0 being installed already. Changing the platform toolset resolved the issue.
For some reason, VS2022 detects a platform toolset called "WindowsApplicationForDrivers10.0(not found)" after installing WDK. That's causing the problem.

swingbozo, 2022-08-19:
Glad I was able to save you some time on this issue.