MicrosoftDocs/windows-driver-docs #2998: Ne fonctionne pas sous Windows 2019/2022

GowaP, 2021-12-17:
L'utilisation de "Bcdedit.exe -set TESTSIGNING ON" ne fonctionne pas avec Windows 2022 Server.
Quel est la nouvelle méthode ?

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7b098cc9-eff8-b24d-824c-99d038196d4c
* Version Independent ID: dce141c3-6962-236a-d418-9ee3b6d48779
* Content: [Loading Test Signed Code - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option)
* Content Source: [windows-driver-docs-pr/install/the-testsigning-boot-configuration-option.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/the-testsigning-boot-configuration-option.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-06:
Hi @gowap, nothing has changed here in WS2022/Win11.  

Note that if you have Memory Integrity / HVCI enabled, you must test-sign the binary (using any self-created test cert). An unsigned binary is not supported.

If that does not apply, would it be possible for you to share the CI logs from `\Windows\System32\winevt\Logs\Microsoft-Windows-CodeIntegrity%4Operational.evtx`?

tedhudek, 2022-01-07:
I added the caveat to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option (see the second Note).  Hope this helps!

Thanks for the feedback :)

closed 2022-01-07 by tedhudek