MicrosoftDocs/windows-driver-docs #2999: EWDK BuildTools language pack?

thxkiwi, 2021-12-17:
Since the EWDK provides a full VS 16 build tools install, what language pack(s) are part of that install?

We use Microsoft's vcpkg (https://github.com/microsoft/vcpkg) to manage dependencies on 3rd party C++ libraries for a service we deploy via INF (required by Windows Update). vcpkg is not able to find the EWDK provided build tools, supposedly because they do not include an English language pack (as suggested by vcpkg contributor here https://github.com/microsoft/vcpkg/issues/22068#issuecomment-997011432)

This creates problems for standardizing a build environment
1. using the EWDK directly
2. using the EWDK within a container

Will the docs team please confirm whether the EWDK installs the English language pack (required by vcpkg) as part of the VS buildtools provided by the EWDK?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-drivers-conceptual-docs**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-07:
Hi @thxkiwi and apologies for the delay on this one.

I'm still waiting for a definitive answer - will circle back here as soon as I have one.

In the meantime, will you let me know if you have resolved this somehow on your own?

tedhudek, 2022-01-15:
@thxkiwi - can you please help me help you by providing the following (guessing here, please correct as needed):

Repro steps: 
1. Open EWDK Build Env.
2. Run a script that ultimately calls vcpkg?

Expected: vcpkg works

Actual: vcpkg fails

Another way to ask: If we do a fix how can we determine if the fix resolves the bug?

thxkiwi, 2022-01-19:
Will get to this when I can. Had to backburner my work on this for other
tasks.

The high level here is that

   - EWDK provides its own compiler
   - If Visual Studio is also installed on the build “host” then vcpkg
   finds only the “host” and does not use the EWDK provided compiler.
   - If no Visual Studio is installed on the host then vcpkg does not find
   a compatible compiler.

*From:* Ted Hudek ***@***.***>
*Sent:* Friday, January 14, 2022 17:11
*To:* MicrosoftDocs/windows-driver-docs <
***@***.***>
*Cc:* Michael Ngarimu ***@***.***>; Mention ***@***.***

*Subject:* Re: [MicrosoftDocs/windows-driver-docs] EWDK BuildTools language
pack? (Issue #2999)

@thxkiwi <https://github.com/thxkiwi> - can you please help me help you by
providing the following (guessing here, please correct as needed):

Repro steps:

   1. Open EWDK Build Env.
   2. Run a script that ultimately calls vcpkg?

Expected: vcpkg works

Actual: vcpkg fails

Another way to ask: If we do a fix how can we determine if the fix resolves
the bug?

thxkiwi, 2022-01-19:
Just wanted to document that incase I can’t get back to this in a
reasonable timeframe.

*From:* Ted Hudek ***@***.***>
*Sent:* Friday, January 14, 2022 17:11
*To:* MicrosoftDocs/windows-driver-docs <
***@***.***>
*Cc:* Michael Ngarimu ***@***.***>; Mention ***@***.***

*Subject:* Re: [MicrosoftDocs/windows-driver-docs] EWDK BuildTools language
pack? (Issue #2999)

@thxkiwi <https://github.com/thxkiwi> - can you please help me help you by
providing the following (guessing here, please correct as needed):

Repro steps:

   1. Open EWDK Build Env.
   2. Run a script that ultimately calls vcpkg?

Expected: vcpkg works

Actual: vcpkg fails

Another way to ask: If we do a fix how can we determine if the fix resolves
the bug?

tedhudek, 2022-01-19:
Hi @thxkiwi, appreciate the reply.  Sorry if I'm not grasping the obvious here, but I have zero context with vcpkg.

Can you give me the specific sequence of steps you took, and then what the specific problem was?

Is it something like this?

1. use vcpkg to install ewdk.
1. try to build service with INF in vcpkg

Expected: driver to build using ewdk compiler
Actual: vcpkg never invokes ewdk compiler, regardless of whether VS is installed on the machine

tedhudek, 2022-01-24:
Closing this out due to inactivity.  Please re-open if still needed

closed 2022-01-24 by tedhudek