MicrosoftDocs/windows-driver-docs #3005: Unclear reference to numbered steps

henke37, 2021-12-21:
The page refers to numbered steps, but the page does not state what those steps are.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 432a9f06-8b6d-e0a6-74e0-8ab82657bf51
* Version Independent ID: 1cdd58c1-2acd-f82a-a75f-401667b6737a
* Content: [Checking Flags in the Device Object - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/checking-flags-in-the-device-object)
* Content Source: [windows-driver-docs-pr/kernel/checking-flags-in-the-device-object.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/checking-flags-in-the-device-object.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-05:
Good eye @henke37, thanks for the report.

It's possible that this was referring to https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/power-irps-for-individual-devices (see the text description of steps 3 and 4, below the image).  These steps mention the driver failing an IRP.  Do think that would make sense?  If so, I can add a link.

Over the years the ordering of pages in this documentation has changed, so it is possible that these pages were next to each other long ago.

henke37, 2022-01-06:
I honestly have no idea if that's the correct page.

tedhudek, 2022-01-07:
Hi @henke37 - I looked a bit closer and I don't think it's that page after all.  So I have no idea what it is referring to!

Is the phrasing blocking you from doing something?  I am tempted to simply remove it.

henke37, 2022-01-07:
Nah, it's not blocking me.

tedhudek, 2022-01-10:
Ok, just removed it - thx for the heads-up!

closed 2022-01-10 by tedhudek