MicrosoftDocs/windows-driver-docs #3008: There is no way of donloading KSStudio Utility

christopherlacapra, 2021-12-23:
Hello,

I am trying to install the KSStudio utility but on its intallation page the instructions are not correct. There is no "Tools for driver developers" element to choose from among the so described list of components. There isn't even a list of components when installing the WDK. 

I am currenctly downloading the WDK from here: https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk
Please update your docs asap.

Regards,
Christopher

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e3f34d2-a2bc-be65-21ff-a8febf1f1ace
* Version Independent ID: f31caf08-8aa4-ac94-5fba-91845ce0e1db
* Content: [KsStudio Utility - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/ksstudio-utility)
* Content Source: [windows-driver-docs-pr/audio/ksstudio-utility.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/ksstudio-utility.md)
* Product: **windows-hardware**
* Technology: **audio**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

christopherlacapra, 2021-12-27:
I found the location of the executable, it is in C:\Program Files (x86)\Windows Kits\10\Tools\x64 for the 64 bits version.
What happened is that the WDK installer installed KSUtility.exe regardless, so I basically had it installed the whole time. 
Which is a good thing, I can work now, but the docs were not very clear on this aspect.

DOMARS, 2022-01-22:
@christopherlacapra - Thanks for this useful documentation input. As you determined, the install directions were out of date. I updated the directions to reflect how the WDK install behaves currently. Thanks for pointing out this docs error, so that we can improve the docs for the next person who is working with the KSStudio utility.

closed 2022-01-22 by DOMARS