MicrosoftDocs/windows-driver-docs #3013: Missing major operating system versions in OSMajorVersion and OSMinorVersion:

Zweater, 2022-01-04:
Missing major operating system versions in OSMajorVersion and OSMinorVersion:

- Windows Server 2016
- Windows Server 2019
- Windows Server 2022
- Windows 11

Do all Windows 10 versions use the same OSMinor version 0? So Windows 10 version 1507 upto Windows 10 version 21H2?
If not, that should be described in the section OSMinorVersion.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9445c977-9f8c-00c6-68fa-b575ab325fcf
* Version Independent ID: 58963a0b-6f4b-b6ee-f454-7e72e68d7bef
* Content: [INF Manufacturer Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section#osminorersion)
* Content Source: [windows-driver-docs-pr/install/inf-manufacturer-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-manufacturer-section.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-06:
Hi @Zweater 

For app compat reasons, all windows 10 versions have a major version of 10 and a minor version of 0.  To differentiate between them in an INF you need to use https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section#buildnumber.

I believe (not 100% sure) that WS2016, WS2019, WS2022 and Win11 are all 10.0 as well - so again, if you are writing an INF for Win10 and later and need to differentiate based on release, again you need to use build number.

Does that help?

tedhudek, 2022-01-06:
I updated https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section#osmajorversion and https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section#osminorversion with the new values.

Thanks again for the feedback!

closed 2022-01-06 by tedhudek

Zweater, 2022-01-07:
Thanks for updating the document!

Regards, Jeroen