MicrosoftDocs/windows-driver-docs #3014: Needs VS 2022 update

markyoung2u, 2022-01-07:
Seems to be for VS 2019 only. 

1. Is https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk also for VS 2022?
2. Step 1 says 2019 - can I load 2022? 
3. Step 2 says "Install Windows 11 SDK (22000.1)" and VS 2022 latest choice [x] I selected was "Windows 11 SDK (10.0.22000.0)" which seems older so I executed step 2 and the ISO told me to remove "Windows Software Development Kit 10.0.22000.194" first, so I did from Programs and Features. The the ISO would run and it installed the exact same version? It was not clear to me at all that being asked to remove it first really meant "You already have it dumbo!"
4. Step 3 installs WDK "Windows Driver Kit - Windows 10.0.22000.1" and says it on the install prompts several times but really adds "C:\Program Files (x86)\Windows Kits\10\bin\10.0.22000.0" and I'm sure because I did it twice and only the date changed.
5.  The next step "Windows 11 EWDK with Visual Studio Build Tools" and says "Note that the Visual Studio major version should match with the version in the EWDK. For example, Visual Studio 2019 works with the EWDK that contain VS16.X build tools" but is not helpful and I stopped here. The Version.txt file in the ISO says "Version co_release.22000.1" - nothing about 2019 or 2022. I don't know if I should keep going or not. No idea if this has VS 16.X tools, it's just a bunch of imprecise statements. I went ahead and it made things worse, sigh.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-drivers-conceptual-docs**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-07:
Hey @markyoung2u my apologies for the confusion - it's an ongoing issue that others have reported as well.  The short story is that VS2022 is not yet supported by the WDK.  I appreciate the details on the discrepancies in the doc page and am keeping this issue open until I get that updated.  It is a known issue internally - we're working on it!

markyoung2u, 2022-01-07:
Is it possible to get Platform Toolset: windowsModeKernelDriver10.0 option installed at this time? (At least it would compile), Also, should I start over with VS 2019, or what a while for VS 2022 to be supported? We are moving from 2017 and I'd prefer waiting a bit to get the latest to remain viable longer, more features, etc

tedhudek, 2022-01-07:
@markyoung2u in #2993 a customer found that he could compile successfully by changing the Platform Toolset from the default to Visual Studio 2022(v143).  My understanding is that won't work in every case though.

Sorry, I don't have any other info for now - will keep this issue open and let you know when that changes.

tedhudek, 2022-01-25:
@markyoung2u see #2988 for the latest update on this.  (I'm still keeping this issue open to track updates to the steps you mention above.)

tedhudek, 2022-02-11:
Thanks again for the feedback @markyoung2u.  I finally had a chance to think through your feedback above, and I made changes to address it in ca30b3673a30b980d3781b700a4287fbfb14eb3f.   See https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk for the live page.

We anticipate publishing the updated kit soon for VS 2022 - stay tuned to the above doc page and #2988 for updates on that.

closed 2022-02-11 by tedhudek