MicrosoftDocs/windows-driver-docs #3015: Entrust Root Certificate Authority - G2 is not available anymore

hnguyenisabel, 2022-01-11:
Dear support,
On this page, before march 2021, under the Cross-Certificate List, the Entrust Root Certificate Authority - G2 was displayed and available for download.
In the actual page, it is removed from the Cross-Certificate List. Therefore I cannot download it anymore.
Is there another way to get this Entrust Root Certificate Authority - G2 again?

Thanks for the help,
Huy Nguyen

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 04c19d51-a948-b3ff-65c2-d8c54291b42f
* Version Independent ID: b5654182-fdcb-72c0-e5ae-536992731e98
* Content: [Cross-Certificates for Kernel Mode Code Signing - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cross-certificates-for-kernel-mode-code-signing)
* Content Source: [windows-driver-docs-pr/install/cross-certificates-for-kernel-mode-code-signing.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/cross-certificates-for-kernel-mode-code-signing.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-11:
Hi @hnguyenisabel,

Thanks for the question.  Can you tell me why you continue to need the cross-certificate?

Please refer to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates#how-do-production-signing-options-differ-by-windows-version, where you'll see "Cross-signing is no longer accepted for driver signing."

hnguyenisabel, 2022-01-12:
Hi @Ted ***@***.***>,

Thanks for the quick response.
We need this cross-certificate because we use an “ENTRUST Root Certification Authority G2.cer” to sign our own CSP (Crypto Service Provider).

We know that starting from Windows 8, it is no longer a requirement that CSPs must be signed.
But since some of our customers still run Windows 7, we still need this cross-certificate.

Thanks for the help,

Huy Nguyen
Isabel Group .SA
Keizerinlaan 13-15 / Boulevard de l’Impératrice 13-15
1000 Brussels, Belgium

Huy Nguyen
Software Development Engineer
t. +32 2 5451 432
m. +32476600604
isabelgroup.eu
 bookmate | clearfacts | codabox | ibanity | isabel 6 | kube | ponto | signhere | zoomit | disclaimer

tedhudek, 2022-01-12:
Hello Huy,

I'm sorry but support for Windows 7 ended in January 2020. Note that you can use the [Lifecycle](https://docs.microsoft.com/en-us/lifecycle/products/) site to look up service dates for different products.

Thanks again for contacting us, and best of luck with this issue.

closed 2022-01-12 by tedhudek