MicrosoftDocs/windows-driver-docs #3021: DbgPrintEx - problems - only show me a 'debug message' on reboot phase

RobertoB74, 2022-01-20:
to be 'strict' ...
1- target machine (virtualized and with net debug options set)
2- host machine (with net debug)
3- the target connect itself to re remote debugger host ...

the problems ....
1. the DbgPrintEx messages are sended and visualized only 'in a reboot phase, in start boot phase'
2. but no when i add and remove my drivers with the hardware manager ...

i retain weird this fact because the DriverEnty and i suppose the DriverUnload functions are however called during the 
hardware installation phase .... 

i have already study the 'complex system' of the 'componentID' and 'mask filters' ... and i think that is setup correctly 

pls someone that could help me to resolve my drama?

i need obviously to trace the various phase during the installation and etc etc ... like each 'debugger techique use'

RobertoB74, 2022-01-21:
one day next .... if the drivers isn't already installed ... on add hardware show me the custom debug trace message ... but only for the first time that i add a new device/driver ... when i call unistal driver and the next times that i will try to reinstall the same driver don't show me anymore nothing ... and in DriverUnload, where there is a debug trace function call don't show me nothing ... :/

RobertoB74, 2022-01-21:
now it's appaire 'all ok/ìworking' there was a 'little error' that did return the DriverEntry function in the middle if(!F(x)) return instead of if (!NTSUCCESS(F(x))) return; 
but however i don't understand why it did has influence on the Debugger System..

RobertoB74, 2022-01-21:
bad return in the middle DriveEntry function has weird influence to the Debugger System messages

closed 2022-01-21 by RobertoB74

DOMARS, 2022-01-21:
@RobertoB74 - This particular channel is monitored by the men and women who write the driver documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to provide support for specific scenarios. 
Your best bet is to use one of the forums where other driver developers and Microsoft engineers can be found. For example the [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html). Another forum for debugging to consider is the OSR Debugger Forum - [OSR's windbg List](https://community.osr.com/categories/windbg)  

For displaying information from a driver, you could consider using WPP Trace as shown in the toaster driver sample - https://github.com/microsoft/Windows-driver-samples/tree/master/general/toaster/toastDrv 

For driver install issues, the debugger echo driver lab mentions the use of the %windir%\inf\setupapi.dev.log - https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-#section-4-install-the-kmdf-echo-driver-sample-on-the-target-system    This log is pretty detailed.

RobertoB74, 2022-01-22:
i am really apologize to you and in the same time tank you too for your information about .... unfortunattly i am a beginner and i don't have found a confortable channels and platform to send my 'beginner questions' and try to find help ...
however ... i have see that the other question ... or are more technical or are more or less the same as mine ... so i hope to have still the opportunity to try to send my informatic damas .... and last but not least .... however this case 'is technically interessant' for the most thecnicall persons and enginerings ... maybe it will be usefull for someone in a future ..... 

have a nice day