MicrosoftDocs/windows-driver-docs #3022: Can't read table on under Host APIs for JavaScript Extensions > 'Root Level'

matwilli, 2022-01-21:
Table seems to be slightly too wide for the page (even when viewed on a wide-screen monitor).  Can't view the entire table without using the horizonal scroll-bar, which is irritating.  Other tables on the page don't have this problem.

[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1f96d161-85bb-9402-d748-a33598beeef9
* Version Independent ID: 01a3c179-7fa1-14b3-2bbc-86d254e176bb
* Content: [Native Debugger Objects in JavaScript Extensions - Debugger Object Details - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/native-objects-in-javascript-extensions-debugger-objects)
* Content Source: [windows-driver-docs-pr/debugger/native-objects-in-javascript-extensions-debugger-objects.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/native-objects-in-javascript-extensions-debugger-objects.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2022-01-21:
Hey Matthew, appreciate the feedback.

I actually tried a few different things to get the table in the Root Level section of that page to not render with a horizontal scroll bar, comparing it to the one in the Memory Functionality section below it that, as you say, does not have this problem.  I tried converting it to a Markdown table, and I tried making the col width values all 25%.  Nothing worked!

So at this point I'm going to close this out, since it is possible (albeit inconvenient, I admit) to use the horizontal scroll bar to see all the content in that table.  If you have any other ideas, feel free to submit a PR :)

Sorry we couldn't get this fixed up!

cc/ @DOMARS

closed 2022-01-21 by tedhudek

matwilli, 2022-01-22:
Thanks Ted, I appreciate the effort.