MicrosoftDocs/windows-driver-docs #3026: where is channel 9?

ShenChenZhou, 2022-01-24:
https://channel9.msdn.com/Shows/Defrag-Tools/Defrag-Tools-186-Time-Travel-Debugging-Advanced

where is this url? It's broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dbe23f42-e182-5876-7324-1abc4338f9c5
* Version Independent ID: c5193b6f-c876-3e20-3ea7-53da98eb4b89
* Content: [Time Travel Debugging - Overview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/time-travel-debugging-overview)
* Content Source: [windows-driver-docs-pr/debugger/time-travel-debugging-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/time-travel-debugging-overview.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2022-01-27:
Hi @ShenChenZhou, thank you for the feedback.  This video is currently offline but we anticipate republishing it sometime in the next couple weeks.  I will keep this issue open and let you know when that happens.  Apologies for any inconvenience this may cause you.

tedhudek, 2022-01-27:
In the meantime, you can try this link instead -- https://sec.ch9.ms/ch9/fa18/d9f6505c-058a-47dd-b69f-4142122bfa18/DefragTools186_mid.mp4

closed 2022-02-10 by tedhudek