MicrosoftDocs/windows-driver-docs #3032: Possible discrepancy between registering & querying for hardware errors?

pastramionry, 2022-02-01:
I noticed when comparing the documentation for querying hardware errors (https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/querying-the-system-event-log-for-hardware-error-events) that the recommendation is to query the Microsoft-Windows-**WHEA-Logger** provider for current versions of Windows (Vista SP1 and later). However, when registering for hardware error events the recommendation is to use the Microsoft-Windows-**Kernel-WHEA**/Errors provider for current versions of Windows (7 and later versions).

Is there a reason for the omission of the WHEA-Logger provider in the registration documentation? The WHEA-Logger provider in the System channel appears to provide way more information than what is found in the Kernel-WHEA/Errors provider on a hardware error.

Apologies if this is a dumb question and thanks for taking the time to look into it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 759a7b35-c5ac-8ead-d483-3417ab3c01ed
* Version Independent ID: 72f27974-7da4-0154-f9c8-e37eacf337ce
* Content: [Registering for Notification of Hardware Error Events - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/registering-for-notification-of-hardware-error-events)
* Content Source: [windows-driver-docs-pr/whea/registering-for-notification-of-hardware-error-events.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/registering-for-notification-of-hardware-error-events.md)
* Product: **windows-hardware**
* Technology: **whea**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

aadriance, 2022-02-01:
Thanks for raising this point. We will work to clarify our recommendations and use cases in the documentation.

To provide some clarity here:
- WHEA-Logger: Is designed for users in desktop scenarios.  It gives a human readable message with the main details of the event so a user that sees the event has some idea on what occurred.
-  Kernel-WHEA/Errors: Is what is recommended for server type scenarios.  The data here isn't immediately human readable, but is an ACPI/UEFI defined Common Platform Error Record (CPER) See: https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/error-records. While the full record requires tooling that's enlightened to understand the formatting due to it's more dynamic nature, it provides a much higher level of detail on the exact specifics of the hardware error event.

Hopefully that provides some clarity on the two different WHEA logs for you to choose the appropriate information source for your use case.

pastramionry, 2022-02-02:
That makes sense and thank you for the additional info. I really appreciate it. Have a good one.

tedhudek, 2022-02-04:
FYI @pastramionry I added the info from this issue to the two pages in question and removed outdated references.  See what you think --

https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/querying-the-system-event-log-for-hardware-error-events
https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/registering-for-notification-of-hardware-error-events

Thanks for raising this!

closed 2022-02-04 by tedhudek