MicrosoftDocs/windows-driver-docs #3040: Is there any page for Windows 11 Bluetooth support?

guliteb, 2022-02-15:
Is there a page to summarize Bluetooth version and profile support in Windows 11?

And the page only listed Version 5.0. However, 5.2 and 5.3 have been announced. When will the page be updated to reflect the support on latest BT version?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 27a0ed90-dff4-f356-1853-de669b0d1540
* Version Independent ID: d846c2ad-9b05-8e61-45f0-19e50734f0f3
* Content: [Bluetooth version and profile support in Windows 10 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/general-bluetooth-support-in-windows)
* Content Source: [windows-driver-docs-pr/bluetooth/general-bluetooth-support-in-windows.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bluetooth/general-bluetooth-support-in-windows.md)
* Product: **windows-hardware**
* Technology: **bluetooth**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2022-02-18:
Hi @guliteb

I'm working on getting the information to update this page to Windows 11.

Thanks for your patience.
 ~ *Mark*

mhopkins-msft, 2022-03-09:
Hi @guliteb,

I've updated the following topic to reflect Windows 11 Bluetooth support: [Bluetooth Version and Profile Support in Windows 11](https://docs.microsoft.com/windows-hardware/drivers/bluetooth/general-bluetooth-support-in-windows)

I've moved the Windows 10 Bluetooth information to this page: [Bluetooth Version and Profile Support in Previous Windows Versions](https://docs.microsoft.com/windows-hardware/drivers/bluetooth/bluetooth-support-in-previous-windows-versions)

Thank you for your helping in making our docs better!
 ~ *Mark*

closed 2022-03-09 by mhopkins-msft