MicrosoftDocs/windows-driver-docs #3047: Do not need to use the same computer to buy and retrieve EV Code Siging certificates

codelux, 2022-02-24:
I don't think this is correct since EV Code Signing certificates must be delivered in hardware devices (either FIPS USB tokens or through HSM). As a result the customer does not need to buy and retrieve the certificate from the same computer/browser:

`[!NOTE] You must use the same computer and browser to buy and retrieve your certificate.`

mhopkins-msft, 2022-02-25:
Hi @codelux,

Can you please tell me the specific page where you found this note?

Thanks,
 ~ *Mark*

codelux, 2022-02-28:
sure, this page https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate

mhopkins-msft, 2022-02-28:
Hi @codelux,

Thank you for pointing this out. I've verified that you do not need to use the same machine and browser to buy and retrieve the certificate. I've updated the topic accordingly.

Thanks for helping to make our docs better!
 ~ *Mark*

closed 2022-02-28 by mhopkins-msft