MicrosoftDocs/windows-driver-docs #3048: Missing background information and explanation

wmjdgla, 2022-02-25:
>  If the caller does not hold this privilege, a traverse check must be done on each directory.

1. `IRP_MJ_CREATE` operates on a single file or directory, so it is unclear what it means by "each directory".
2. Why must a traverse check be done? What are the implications of not doing it?

Then in the sample code:

 ```
        //
        // this changes the AccessState
        //
        (void) SeAppendPrivileges(AccessParams.AccessState, Privileges );
```
It is unclear why a traverse check would require changing the `AccessState`.

Likewise the rest of the sample code does not appear to have any relation to `SeChangeNotifyPrivilege` or traversal privilege. Overall the article makes little sense and is pretty bewildering.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 312d414a-e95d-b15b-a93f-c6e5a3698ba9
* Version Independent ID: 62e9b552-e83f-61f8-2d28-07c85aa91d4e
* Content: [Checking for Traverse Privilege on IRP_MJ_CREATE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/checking-for-traverse-privilege-on-irp-mj-create)
* Content Source: [windows-driver-docs-pr/ifs/checking-for-traverse-privilege-on-irp-mj-create.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/checking-for-traverse-privilege-on-irp-mj-create.md)
* Product: **windows-hardware**
* Technology: **ifs**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2022-03-17:
Thanks for filing this issue, @wmjdgla. While it was accurate, there were definitely some big gaps in details. I've updated the page and code snippet to answer your questions and bring the various elements and concepts together.

closed 2022-03-17 by lorihollasch

wmjdgla, 2022-04-15:
@lorihollasch Thanks for the update, I understand it much better now.
I noticed that the line:

> `if (!checkOnly) {`

Is missing the closing brace, so it's unclear what the scope of the subsequent lines of code is. Should I create a new issue for this?

lorihollasch, 2022-04-22:
@wmjdgla, I've added the missing brace - thanks for catching that. Glad the page update helped!