MicrosoftDocs/windows-driver-docs #3054: Document how to use WaitCompletionPacket

DemiMarie, 2022-03-05:
The only efficient way to wait for a large number of waitable objects is to use the undocumented functions NtCreateWaitCompletionPacket, NtAssociateWaitCompletionPacket, and NtCancelWaitCompletionPacket.  These functions are used internally by the Windows thread pool implementation, but are not documented for use in user code.  The result is that event loops that are based on I/O Completion Ports do not support waitable objects.

tedhudek, 2022-03-07:
Hi @DemiMarie, thank you for contacting us.  I just searched both the WDK and SDK headers (from the Win11 kits) and did not see the routines you list.  Our general rule is if it's not in a header that we ship, we don't document it.

You could try the forums for further help -- are you writing an app or a driver?

DemiMarie, 2022-03-07:
> Hi @DemiMarie, thank you for contacting us. I just searched both the WDK and SDK headers (from the Win11 kits) and did not see the routines you list. Our general rule is if it's not in a header that [...]

I want to be able to wait for various waitable objects in a userspace application that uses I/O Completion Ports.  Using another thread is inefficient.

closed 2022-03-08 by tedhudek

DemiMarie, 2022-03-08:
> I'd suggest using the user-mode thread pool APIs. You can find more info at https://docs.microsoft.com/en-us/windows/win32/procthread/thread-pool-api.
> 
> Thanks for reaching out and best of luck!

Sadly the thread pool APIs are not practical to integrate into existing event loops.  WaitCompletionPacket, on the other hand, works fantastically in such event loops, and most portable applications do have their own event loops.