MicrosoftDocs/windows-driver-docs #3060: Only initial package in Targeting section can be resold?

belle-hsu, 2022-03-11:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4b598a4f-b917-cdc9-6a85-c5d0e9cbd91e
* Version Independent ID: 23a9abff-5dd9-ebf9-1044-b2a5c2e9a453
* Content: [Share a driver with a partner - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/sharing-drivers-with-your-partners)
* Content Source: [windows-driver-docs-pr/dashboard/sharing-drivers-with-your-partners.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/sharing-drivers-with-your-partners.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

belle-hsu, 2022-03-11:
We would like to confirm if only initial driver package in Targeting section can be resold when we share a driver with a partner?

belle-hsu, 2022-03-15:
I just received the feedback from MSFT Dev Center, appreciate that, so that means only initial driver package can be resold, correct?

Comment from [DEVAB.GNRL.WW.00.EN.TPR.BOG.DN.T01.SPT.SG.WB@css.one.microsoft.com]
==============================================================================
Hello There,
Hope this email finds you well.
Thank you for contacting Microsoft’s Developer Support Center.
We would like to kindly inform you that the DUA shell would only be available for Initial submissions and not derived\resold submissions (DUA = Driver Update Acceptable. On the portal these type of submissions are called a Derived submission).
When a “Derived” submission is shared to a partner in the Partner Center, the receiver is unable to create another derived submission from that submission. The lack of a DUA Shell button means that the Author  shared out a derived submission. This is the only way a receiver will know what the Author did, as we do not show this on the receiver’s end.
Please review the following article from the official Microsoft's documentation for more information on the management of your Hardware submissions:
•	https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/manage-your-hardware-submissions
If you have any questions or further request do not hesitate to contact us.
 
Wishing you a lovely day!
Kind regards,
Viktor J.   
Your Partner at Microsoft Dev Center
Support Hours: Monday-Friday, 8 a.m.-8 p.m. EDT
https://developer.microsoft.com/en-us/windows/support
Twitter: [@WindowsDev](https://nam06.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.twitter.com%2Fwindowsdev&data=04%7C01%7CDEVAB.GNRL.WW.00.EN.TPR.BOG.DN.T01.SPT.SG.WB%40css.one.microsoft.com%7C89232402cf4746ceed0608d9fdeec9ef%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637820024345163847%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=27zP6%2F78YY%2Fj5bFKV7T4Bqz5qY6WsbjGG1OOgs7xq%2F0%3D&reserved=0)
CASE ID: 1536730768

mhopkins-msft, 2022-03-17:
Hi @belle-hsu,

Thanks for reaching out to our team. This particular mechanism is monitored by the men and women who write the driver documentation posted on doc.microsoft.com. Unfortunately, we don’t have the resources to help troubleshoot driver development issues sent our way.  
 
Your best bet is to use one or both of the two really good forums where other driver developers and Microsoft engineers can be found. The first is [Microsoft Q&A](https://docs.microsoft.com/en-us/answers/products/). The other is the [OSR NTDEV forum](https://community.osr.com/categories/ntdev). You can post your question there, and like-minded developers usually give feedback fairly quickly. 

I'm glad you received an answer from Viktor with Microsoft’s Developer Support Center. You could also follow up with them.
 
Again, thanks for reaching out. Best of luck with this issue.

 ~ *Mark*

closed 2022-03-17 by mhopkins-msft