MicrosoftDocs/windows-driver-docs #3067: FLT_PREOP_SYNCRHONIZE can return in another thread when draining

rodwiddowson, 2022-03-15:
[Enter feedback here]
Hi @lorihollasch ,

I'm not going to even try a PR for this one but I'll observe that 

> The filter manager calls the minifilter driver's postoperation callback routine in the same thread context as the preoperation callback,

Is only true "up to a point".  During instance (or maybe filter manager) teardown I have seen returns come in different threads (as per below)

```
XXX!XXPostCreate+0x3b89 
FLTMGR!FltvPostOperation+0xb4
FLTMGR!DrainCompletionNode+0x1b9
FLTMGR!FltpDrainActiveCompletions+0x86f0
FLTMGR!FltpFreeInstance+0x100
FLTMGR!FltpCommonDetachVolumeWorker+0x17
```

Of course this actually means that you should not hold resources between pre & post calls - or if you do you should squirrel away the threadId so you can release on behalf of the thread) but that doesn't solve the ApcDisable count problem.

Of course the stack about is actually covered by this text

>  If a minifilter driver has registered preoperation and postoperation callback routines for IRP_MJ_CREATE operations, the post-create > callback routine is called at IRQL = PASSIVE_LEVEL, in the same thread context as the pre-create callback routine.

But the net result is the same.

I have *no idea* how to describe this behavior without destroying the whole concept of `FLT_PREOP_SYNCHRONIZE` as a useful thing.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6d4bd0c9-72cf-d978-dbdb-8b70baf1360c
* Version Independent ID: d972a42c-5a70-0f71-893f-053a93c93712
* Content: [Returning FLT_PREOP_SYNCHRONIZE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/returning-flt-preop-synchronize)
* Content Source: [windows-driver-docs-pr/ifs/returning-flt-preop-synchronize.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/returning-flt-preop-synchronize.md)
* Product: **windows-hardware**
* Technology: **ifs**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2022-06-07:
@lorihollasch Nothing personal here but I just got bitten by this wretched restriction again with "a possible deadlock in verifier" which I cannot debug because I have passed ownership of the ERESOURCE from the thread to a context in my pre-call (so that I can safely drop it in the post).

When you document this it would be good to outline winning strategies for avoiding the issue.  Pseudocode follows:

```
ERESOURCE_THREAD GetResourceThread(PVOID Context) {
    ULONG_PTR ptr = (ULONG_PTR) Context;
    ptr |=3;
    return ptr;
}

PVOID GetResourceThreadPtr(PVOID Context) {
    return (PVOID)GetResourceThread(Context);
}

PRE:
   ExAcquireResourceExclusiveLite(&context->Lock, TRUE);
   ExSetResourceOwnerPointer(&context->Lock, GetResourceThreadPtr(context));
#if DBG > 0
   context->LockOwner = PsGetCurrentThread(); 
#endif
   return FLT_PREOP_SYNCRHONIZE; // Even tho' we don't own the lock we have to syncrhonize or self-deadlock on the resource

POST:
#if DBG > 0
   context->LockOwner = NULL;
#endif
   // Do stuff to shut up verifier
   ExReleaseResourceForThreadLite(&context->Lock, GetResourceThread(context)
```

cgallred, 2022-07-09:
Hi @rodwiddowson, sorry for the delay getting back to you. You're right that Filter Manager does not sync back to the original thread if the filter is draining. But you're also right that it means you should not hold resources between pre- and post- calls, same as you shouldn't hold a resource across an I/O call That's not safe, as it can result in deadlocks.

Neal and I talked about `FLT_PREOP_SYNCHRONIZE` and, funny thing, we couldn't really come up with a compelling reason for it to exist. :-/ Aside from giving a filter writer a way to hold a resource across pre- and post-, which they should not do, we couldn't think of a scenario where it would be useful.

lorihollasch, 2022-07-20:
Thanks as always for flagging yet another gnarly issue, @rodwiddowson! And thanks to @cgallred for your insight into this issue. I've updated the verbiage regarding thread sync. I also added a note that resources should not be held across pre- and post- calls. Hopefully that will help others from getting bit.

closed 2022-07-20 by lorihollasch