MicrosoftDocs/windows-driver-docs #3068: Missing "Submit new hardware"

charvatjiri, 2022-03-21:
[Enter feedback here]
>> Sign in to the [Partner Center](https://partner.microsoft.com/dashboard), and then select Submit new hardware.
Few people tried this in our organization but nobody saw any link or menu item named "Submit new hardware". Maybe it is visible only under some conditions or it is somewhere else. It would be worth to note where / when the item "Submit new hardware" is available. Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 16149520-06cf-5257-0e53-0aeda1719b2a
* Version Independent ID: 63b8a418-0c66-c9f9-321f-f954543ec452
* Content: [Create a new hardware submission - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-hardware-submission)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-hardware-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-hardware-submission.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2022-03-24:
Hi @charvatjiri,

Your company must first be registered to submit new hardware.

https://partner.microsoft.com/en-us/dashboard/Registration/Hardware

Have you done this step yet?

 ~ *Mark*

charvatjiri, 2022-03-27:
Hello @mhopkins-msft,
yes I confirm it is now Ok.

Best regards
Jiri

mhopkins-msft, 2022-03-28:
Great! I'm glad to hear that you are unblocked.