MicrosoftDocs/windows-driver-docs #3071: Installation instructions don't provide installation instructions.

fmokren, 2022-03-22:
The heading says, "Installing and configuring Application Verifier", but I don't see any instructions on installing Application Verifier. Where to download, etc.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 49bfe16d-c09b-2dd9-dc97-16d57bff993c
* Version Independent ID: 49bfe16d-c09b-2dd9-dc97-16d57bff993c
* Content: [Application Verifier - Testing Applications - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/application-verifier-testing-applications)
* Content Source: [windows-driver-docs-pr/devtest/application-verifier-testing-applications.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/application-verifier-testing-applications.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2022-03-25:
@fmokren - Thanks for this input. I updated the section in this topic to just say configuration and I added a new Installation section in the main page - https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/application-verifier?branch=domars-qemu#installing-appverifier. Thanks again for the feedback on the docs and good luck with your use of App Verifier.

closed 2022-03-25 by DOMARS

fmokren, 2022-03-25:
Thank you. App Verifier helped me kind a seriously bad bug.