MicrosoftDocs/windows-driver-docs #3072: How to disable Basic Connect Extensions

ksh1ng, 2022-03-24:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ef4e8d2f-8bd3-eae7-ff21-f369672e07e9
* Version Independent ID: 3d5fa26f-5e9b-328d-84c7-f3176ecf1e68
* Content: [MB LTE Attach Operations - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-lte-attach-operations)
* Content Source: [windows-driver-docs-pr/network/mb-lte-attach-operations.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-lte-attach-operations.md)
* Product: **windows-hardware**
* Technology: **network**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-03-24:
Hi @ksh1ng,

Basic Connect Extensions refers to a group of CIDs at the MBIM interface. Such a group is referred to as a Device Service. There is no way to disable the device service of Basic Connect Extensions at the MBIM interface from the MBIM device side.

Can you please clarify your question and provide some context on what you are trying to achieve? 

Thanks.

aviviano, 2022-03-31:
Closing due to inactivity.

closed 2022-03-31 by aviviano