MicrosoftDocs/windows-driver-docs #3074: This article says Windows Drivers can be installed on the HoloLens but there is nowhere a guide how to do this.

muenchris, 2022-03-24:
[Enter feedback here]
We are trying to get a driver for a USB Dongle installed on the HL2. This article looks promising if we write the driver as a "Windows Driver" but I cannot find anywhere a way to install a Windows Driver on the HoloLens. Can you either point at a page how to install a Windows Driver on HoloLens or create a new page?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e7dd9475-4e84-c0b0-a4b0-4fd3379e13ba
* Version Independent ID: f554aafc-0f0f-f2f3-4b1d-9f4cf05b2db8
* Content: [Getting Started with Windows Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-windows-drivers)
* Content Source: [windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md)
* Product: **windows-hardware**
* Technology: **develop**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-03-25:
Hi @muenchris,

Thanks for reaching out.

There is a list of HoloLens 2 supported USB devices at https://docs.microsoft.com/en-us/hololens/hololens-connect-devices#hololens-2-connect-usb-c-devices.

Anything not listed there isn’t supported. If the higher-level use case is something like a USB smart card reader or similar device, there are guidelines at https://docs.microsoft.com/en-us/hololens/security-limiting-password-use#signing-in-from-another-device.

I will keep this issue open until I can add these links to the page you referenced.

muenchris, 2022-03-25:
Thanks for your fast reply. I am familiar with that page and during my research I stumbled across your page. It kind of eludes readers that one could install a "UWP Windows Driver" on the HoloLens but it does not show a way how to.
Traditional kernel and "Desktop Drivers" cannot be installed and make sense, but anything I can install with an APPX including these "universal drivers" could work - at least that was my conclusion I drew from this documentation page. If this is not the case, you might want to note this on this page that even "(Universal/User-Mode) Windows Driver" cannot be installed on HLs.

closed 2022-05-17 by tedhudek