MicrosoftDocs/windows-driver-docs #3075: cannot GET https://manage.devcenter.microsoft.com/v2.0/my/hardware/products/,statusCode = 500

zyc0327, 2022-03-25:
[Enter feedback here]
GET URI=https://manage.devcenter.microsoft.com/v2.0/my/hardware/products/ with Postman,
response=
{
    "statusCode": 500,
    "message": "Internal server error",
    "activityId": "976a2ba3-a2d1-469f-8ea7-a51ac9df4c31"
}
How To Get A Currect Response?
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40f09988-3ba3-0ce2-9952-b11ddd66aff1
* Version Independent ID: dd0d280c-f6a0-4a07-7f86-42eafaf6087b
* Content: [Get all products - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-all-products)
* Content Source: [windows-driver-docs-pr/dashboard/get-all-products.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-all-products.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2022-03-28:
Hi @zyc0327,

The example code on the page shows that you must include your access token in the request:

```
GET https://manage.devcenter.microsoft.com/v2.0/my/hardware/products/ HTTP/1.1
Authorization: Bearer <your access token>
```

Did you include this?

Also, did you [complete the prerequisites for using the Microsoft Hardware API](https://docs.microsoft.com/windows-hardware/drivers/dashboard/dashboard-api#complete-the-prerequisites-for-using-the-microsoft-hardware-api)?

 ~ *Mark*

zyc0327, 2022-03-29:
Yes,I could generate a Bearer successfully And Bearer is already included in the request.

closed 2022-03-30 by zyc0327