MicrosoftDocs/windows-driver-docs #3084: Cannot commit a product submission

zyc0327, 2022-03-30:
From the Document

> https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/commit-a-product-submission
I'm trying to Commit a product submission by 

> https://manage.devcenter.microsoft.com/v2.0/my/hardware/products/{productID}/submissions/{submissionID}/commit
it returns a null response,status code = 202,I think it works.
But when I'm trying to get the status of the submission
> https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-submission
It returns
[image]
and the web page shows
[image]
How can I commit a submission Successfully？

zyc0327, 2022-03-30:
By the way,
From the document Step 7:

> https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/manage-product-submissions

> _To confirm the submission status, review the commitStatus value in the response body. This value should change from CommitReceived to CommitComplete if the request succeeds or to CommitFailed if there are errors in the request. If there are errors, the error field contains further details about the error._

There is no error field in the response

mhopkins-msft, 2022-03-31:
Hi @zyc0327,

I've updated the topic to remove the mention of error details.

Thank you for your feedback. I appreciate you helping us make the docs better!

 ~ *Mark*