MicrosoftDocs/windows-driver-docs #3089: no certmgr.exe located in Windows 10 and Server 2016. please correct info on this page

anon193465, 2022-03-30:
Hi,

certmgr.exe does not exist on Windows 10 and Server 2016. please advise if this is in error and please provide clean and disambiguous information.

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d4990dff-7776-9e5e-36e0-7ca122e6b9cc
* Version Independent ID: 048b2cdf-c80b-59bf-4f8a-deb812585886
* Content: [Using CertMgr to Install Test Certificates on a Test Computer - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-certmgr-to-install-test-certificates-on-a-test-computer)
* Content Source: [windows-driver-docs-pr/install/using-certmgr-to-install-test-certificates-on-a-test-computer.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-certmgr-to-install-test-certificates-on-a-test-computer.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2022-04-01:
@anon193465     Certmgr.exe is there in windows 10 and windows too, just now i checked on windows 11.

anon193465, 2022-04-01:
Hi,

Please search your hard drive with this command: **dir c:\*certmgr.exe /s** and let me know where the EXE is. 

However, search **dir c:\*certmgr* /s** and you will find:

**C:\Windows\System32**

04/09/2021  06:49 AM         2,237,440 certmgr.dll
12/07/2019  02:09 AM            63,070 **certmgr.msc**
               2 File(s)      2,300,510 bytes

**C:\Windows\SysWOW64**

12/07/2019  02:10 AM            63,070 **certmgr.msc**

Where is your EXE file located?

-Jason

tedhudek, 2022-04-11:
Hi @anon193465 sorry for the delay on this one.

On Windows 11, in PowerShell, when I type `gcm certmgr` I get:

```
CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Application     certmgr.msc                                        0.0.0.0    C:\WINDOWS\system32\certmgr.msc
```

So maybe we need to change the .exe to .msc on https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-certmgr-to-install-test-certificates-on-a-test-computer?  Let me know if that works for you and I can update the doc page.

Thanks for the report.

anon193465, 2022-04-11:
> Hi @anon193465 sorry for the delay on this one.
>
> On Windows 11, in PowerShell, when I type `gcm certmgr` I get:
>
> ```
> CommandType     Name                                               Version [...]

I'm trying to learn about command-line CERTMGR.MSC (or whatever it is) so I hope this info gets corrected and expanded upon. Information like how to add certificates to and from the different trust locations would be great. It's a big topic that I think should be greatly expanded upon with more information.

tedhudek, 2022-05-17:
Closing out for now.  We are planning future updates to the code signing docs but I do not currently have an ETA for that work.

closed 2022-05-17 by tedhudek