MicrosoftDocs/windows-driver-docs #3093: WindowsVersionList Outdated

druske, 2022-04-04:
I suspect that the WindowsVersionList table may be outdated; the latest thing on there is 10_19H1_x64

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0afbc04b-349a-57ed-eeb0-3d2cd28e359c
* Version Independent ID: de61f684-31d7-2de5-d66c-a3319f5198d9
* Content: [Inf2Cat - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf2cat)
* Content Source: [windows-driver-docs-pr/devtest/inf2cat.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf2cat.md)
* Product: **windows-hardware**
* Technology: **devtest**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2022-04-12:
@DOMARS I am going to  change the location path.

DOMARS, 2022-05-06:
@druske - Thanks for filing this issue and pointing out the need to update the version list for current day versions of Windows. This list has been updated, including Windows 11.

closed 2022-05-06 by DOMARS