MicrosoftDocs/windows-driver-docs #3094: How to disable Basic Connect Extensions from Host(OS) Side

ksh1ng, 2022-04-06:
[image]
As follow-up from #3072, the problem we're encountering is that **the LTE attach configuration of our MBIM device will be set by the MBIM interface, resulting in the behavior of attaching with an unexpected APN**. For example, original attach_pdn_list = { 2, 1}, when SIM inserted, attach_pdn_list changed to  { 2, 2, 2 } by **MS_LTE_ATTACH_CFG (SET)**, which is  not what we want.

The attached files are USB logs and a screenshot of the MS_LTE_ATTACH_CFG (SET).

ksh1ng, 2022-04-07:
Hi @aviviano,
As the product team mentioned "Windows OS uses the MBIM interface to change LTE attach configuration only if there is some LTE attach configuration from a source with higher priority than those in the MBIM device." **May I know the detailed information about this priority (e.g., priority order of all sources)?**
For other technical questions, I will reach out to Mobile Broadband Discussion, thanks for your information.

Respectfully.

ksh1ng, 2022-04-07:
Hi @aviviano,
Oops, I encounter an issue when email to [mbtalk@microsoft.com](mailto:mbtalk@microsoft.com). Is it possible to solve it? Thanks!
[image]

aviviano, 2022-04-07:
Hi @ksh1ng, can you please provide an email I can pass on to the product team? Thanks.

ksh1ng, 2022-04-09:
Hi @aviviano,
opompm841218@gmail.com 

Thanks & Respectfully.

ksh1ng, 2022-04-14:
Hi @aviviano,
Is there any update on that?
Thanks & Respectfully.

aviviano, 2022-04-14:
Sorry for the delay @ksh1ng. I'm waiting on the team to add you to the email allow list. I'll let you know as soon as I hear back from them.

aviviano, 2022-04-15:
Hi @ksh1ng,
The team isn't sure if we can add an external person to the allow list for mbtalk@microsoft.com. While we are waiting on an answer, you can try posting your questions on [Hardware and Application Developers on Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html) or the OSR [NTDEV forum](https://community.osr.com/categories/ntdev). Unfortunately, I am unable to help further with your specific scenario. I will contact you if the team manages to get you access to the mbtalk email.

aviviano, 2022-04-25:
Hi @ksh1ng,
Unfortunately we can't provide you access to mbtalk@microsoft.com because it is for internal Microsoft employees only. My apologies for any confusion. Please try forums I mentioned. The developers are quite active there. Best of luck solving your issue.